gotthardp/lorawan-server #238: Support of Class B

pdl86, 2017-11-20:
Hello 
do you have any plan to support Lorawan Class B device ? 
we have a need to "wake-up" a device while "sleeping". The idea is then to use Class B mode , with a listening windows .
thanks in advance 
regards
Philippe

gotthardp, 2017-11-21:
Hello. I have no plans (yet) to implement class B. I am even not sure it can be implemented using the existing packet_forwarder.

gotthardp, 2017-11-21:
Or you can perform a thorough analysis of the lorawan spec and the packet-forwarder protocol and derive the absolute minimum requirements for class B support. I may be wrong when I say It cannot be done, but I have no time at this moment to analyze it further. (Though just implementing the min requirements might be easier.)

pdl86, 2017-11-21:
will do
I will let you know
regards
philippe