gotthardp/lorawan-server #240: Tiny erlang ?

hallard, 2017-11-24:
Hi there,
Is there a way to install erlang with minimal setup ?
On fresh Debian Strech without any X system, it requires 186 packages and 579MB of disk space.

````
root@eeebox:~# apt-get install erlang
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  adwaita-icon-theme at-spi2-core ca-certificates-java dconf-gsettings-backend dconf-service default-jre-headless
  erlang-asn1 erlang-base erlang-common-test erlang-corba erlang-crypto erlang-debugger erlang-dev erlang-dialyzer
  erlang-diameter erlang-edoc erlang-eldap erlang-erl-docgen erlang-et erlang-eunit erlang-examples erlang-gs
  erlang-ic erlang-ic-java erlang-inets erlang-jinterface erlang-megaco erlang-mnesia erlang-mode erlang-observer
  erlang-odbc erlang-os-mon erlang-parsetools erlang-percept erlang-public-key erlang-reltool erlang-runtime-tools
  erlang-snmp erlang-src erlang-ssh erlang-ssl erlang-syntax-tools erlang-tools erlang-typer erlang-wx erlang-xmerl
  fontconfig fontconfig-config fonts-dejavu-core glib-networking glib-networking-common glib-networking-services
  gnome-icon-theme gsettings-desktop-schemas gtk-update-icon-cache hicolor-icon-theme java-common javascript-common
  libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3 libavahi-common-data libavahi-common3
  libcairo-gobject2 libcairo2 libcolord2 libcroco3 libcups2 libdatrie1 libdconf1 libdrm-amdgpu1 libdrm-intel1
  libdrm-nouveau2 libdrm-radeon1 libdrm2 libegl1-mesa libepoxy0 libfontconfig1 libfontenc1 libgail-common libgail18
  libgbm1 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa libglib2.0-0
  libglib2.0-data libglu1-mesa libgraphite2-3 libgtk-3-0 libgtk-3-bin libgtk-3-common libgtk2.0-0 libgtk2.0-bin
  libgtk2.0-common libharfbuzz0b libice6 libjbig0 libjpeg62-turbo libjs-jquery libjs-jquery-metadata
  libjs-jquery-tablesorter libjson-glib-1.0-0 libjson-glib-1.0-common liblcms2-2 libllvm3.9 libltdl7 libnotify4
  libnspr4 libnss3 libodbc1 libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpciaccess0 libpcsclite1
  libpixman-1-0 libproxy1v5 librest-0.7-0 librsvg2-2 librsvg2-common libsctp1 libsensors4 libsm6 libsoup-gnome2.4-1
  libsoup2.4-1 libtcl8.6 libthai-data libthai0 libtiff5 libtk8.6 libtxc-dxtn-s2tc libutempter0 libwayland-client0
  libwayland-cursor0 libwayland-egl1-mesa libwayland-server0 libwxbase3.0-0v5 libwxgtk3.0-0v5 libx11-xcb1 libxaw7
  libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0 libxcb-shape0 libxcb-shm0 libxcb-sync1
  libxcb-xfixes0 libxcomposite1 libxcursor1 libxdamage1 libxfixes3 libxft2 libxi6 libxinerama1 libxkbcommon0 libxmu6
  libxpm4 libxrandr2 libxrender1 libxshmfence1 libxss1 libxt6 libxtst6 libxv1 libxxf86dga1 libxxf86vm1
  notification-daemon openjdk-8-jre-headless shared-mime-info tcl tcl8.6 tk tk8.6 x11-common x11-utils xbitmaps
  xdg-user-dirs xterm
Suggested packages:
  default-jre erlang-manpages erlang-doc xsltproc fop apache2 | lighttpd | httpd colord cups-common gvfs
  liblcms2-utils libmyodbc odbc-postgresql tdsodbc unixodbc-bin pcscd librsvg2-bin lksctp-tools lm-sensors libnss-mdns
  fonts-dejavu-extra fonts-ipafont-gothic fonts-ipafont-mincho fonts-wqy-microhei fonts-wqy-zenhei fonts-indic
  tcl-tclreadline mesa-utils xfonts-cyrillic
The following NEW packages will be installed:
  adwaita-icon-theme at-spi2-core ca-certificates-java dconf-gsettings-backend dconf-service default-jre-headless
  erlang erlang-asn1 erlang-base erlang-common-test erlang-corba erlang-crypto erlang-debugger erlang-dev
  erlang-dialyzer erlang-diameter erlang-edoc erlang-eldap erlang-erl-docgen erlang-et erlang-eunit erlang-examples
  erlang-gs erlang-ic erlang-ic-java erlang-inets erlang-jinterface erlang-megaco erlang-mnesia erlang-mode
  erlang-observer erlang-odbc erlang-os-mon erlang-parsetools erlang-percept erlang-public-key erlang-reltool
  erlang-runtime-tools erlang-snmp erlang-src erlang-ssh erlang-ssl erlang-syntax-tools erlang-tools erlang-typer
  erlang-wx erlang-xmerl fontconfig fontconfig-config fonts-dejavu-core glib-networking glib-networking-common
  glib-networking-services gnome-icon-theme gsettings-desktop-schemas gtk-update-icon-cache hicolor-icon-theme
  java-common javascript-common libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3
  libavahi-common-data libavahi-common3 libcairo-gobject2 libcairo2 libcolord2 libcroco3 libcups2 libdatrie1 libdconf1
  libdrm-amdgpu1 libdrm-intel1 libdrm-nouveau2 libdrm-radeon1 libdrm2 libegl1-mesa libepoxy0 libfontconfig1
  libfontenc1 libgail-common libgail18 libgbm1 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common libgl1-mesa-dri
  libgl1-mesa-glx libglapi-mesa libglib2.0-0 libglib2.0-data libglu1-mesa libgraphite2-3 libgtk-3-0 libgtk-3-bin
  libgtk-3-common libgtk2.0-0 libgtk2.0-bin libgtk2.0-common libharfbuzz0b libice6 libjbig0 libjpeg62-turbo
  libjs-jquery libjs-jquery-metadata libjs-jquery-tablesorter libjson-glib-1.0-0 libjson-glib-1.0-common liblcms2-2
  libllvm3.9 libltdl7 libnotify4 libnspr4 libnss3 libodbc1 libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0
  libpciaccess0 libpcsclite1 libpixman-1-0 libproxy1v5 librest-0.7-0 librsvg2-2 librsvg2-common libsctp1 libsensors4
  libsm6 libsoup-gnome2.4-1 libsoup2.4-1 libtcl8.6 libthai-data libthai0 libtiff5 libtk8.6 libtxc-dxtn-s2tc
  libutempter0 libwayland-client0 libwayland-cursor0 libwayland-egl1-mesa libwayland-server0 libwxbase3.0-0v5
  libwxgtk3.0-0v5 libx11-xcb1 libxaw7 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0
  libxcb-shape0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0 libxcomposite1 libxcursor1 libxdamage1 libxfixes3 libxft2
  libxi6 libxinerama1 libxkbcommon0 libxmu6 libxpm4 libxrandr2 libxrender1 libxshmfence1 libxss1 libxt6 libxtst6
  libxv1 libxxf86dga1 libxxf86vm1 notification-daemon openjdk-8-jre-headless shared-mime-info tcl tcl8.6 tk tk8.6
  x11-common x11-utils xbitmaps xdg-user-dirs xterm
0 upgraded, 186 newly installed, 0 to remove and 0 not upgraded.
Need to get 145 MB of archives.
After this operation, 579 MB of additional disk space will be used.
Do you want to continue? [Y/n] n
````

gotthardp, 2017-11-24:
Hello. You don't need to install everything. Perhaps I should describe the minimum list....

Please try `erlang erlang-compiler erlang-syntax-tools erlang-crypto erlang-inets erlang-asn1 erlang-public-key erlang-ssl erlang-mnesia erlang-os-mon` first and let me know if it fails.

hallard, 2017-11-24:
Petr
I'll tried this next time, I think it will also need erlang-cowbow ;-)

FYI I tried with minimal (not these one but lot of other) 2 days ago, and didn't work. But as installation of lorawan-server with `dpkg -i` require erlang, install seem to check all erlang package and if one missing install don't go further.

May be an idea would be for testing
- removing erlang dependencies from dpkg package so installer install lorawan-server without any complains
- start by hand lorawan-server until it works, installing erlang-* needed package one by one to isolate the needed ones ?

gotthardp, 2017-11-24:
Cowboy is included with the server, so you don't need to install it.

Task for me: to update the .dpk dependencies, so it include only the minimum required. Users should not be required to install all Erlang packages.

hallard, 2017-11-24:
Thank you very much

gotthardp, 2018-01-11:
I just fixed https://github.com/gotthardp/lorawan-server/blob/devel/doc/Installation.md to reflect the minimum dependencies.

closed 2018-01-11 by gotthardp

hallard, 2018-01-13:
Petr,
looks like .deb package refers to erlang (the full one) not erlang-base, so installing only erlang package mentioned does not works when installing from .deb built with `make deb`

```

loragw:~/lorawan-server$ make release

...
bla bla bla
...

===> Resolved 'lorawan-server'-0.5.0
===> Dev mode enabled, release will be symlinked
===> release successfully created!

loragw:~/lorawan-server$ make dpkg
./scripts/dpkg-deb/build-deb
dpkg-deb: building package 'lorawan-server' in '/home/loragw/lorawan-server/scripts/dpkg-deb/../../_build/default/rel/lorawan-server/lorawan-server_0.5.0_all.deb'.

loragw:~/lorawan-server$ sudo dpkg -i _build/default/rel/lorawan-server/lorawan-server_0.5.0_all.deb 
Selecting previously unselected package lorawan-server.
(Reading database ... 40980 files and directories currently installed.)
Preparing to unpack .../lorawan-server_0.5.0_all.deb ...
Unpacking lorawan-server (0.5.0) ...
dpkg: dependency problems prevent configuration of lorawan-server:
 lorawan-server depends on erlang (>= 19.0-1); however:
  Package erlang is not installed.

dpkg: error processing package lorawan-server (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 lorawan-server
```

gotthardp, 2018-01-13:
Oh no, stupid me! Apologies. I apparently didn't commit the right file. Please try again now.

hallard, 2018-01-22:
Tried on a fresh install, works!

Thanks