gotthardp/lorawan-server #241: ack_lost

webfrank, 2017-11-24:
Hi,
  after few hours of regular working I got continuos "ack_lost" errors from the gateway but uplink/downlink works perfectly and from log I have no errors.

This is the gateway log
```Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: ### [UPSTREAM] ###
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # RF packets received by concentrator: 1
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # RF packets forwarded: 1 (14 bytes)
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # PUSH_DATA datagrams sent: 2 (304 bytes)
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # PUSH_DATA acknowledged: 0.00%
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: ### [DOWNSTREAM] ###
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # PULL_DATA sent: 3 (100.00% acknowledged)
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # PULL_RESP(onse) datagrams received: 1 (174 bytes)
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # RF packets sent to concentrator: 1 (14 bytes)
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # TX errors: 0
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # TX rejected (collision packet): 0.00% (req:1, rej:0)
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # TX rejected (collision beacon): 0.00% (req:1, rej:0)
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # TX rejected (too late): 0.00% (req:1, rej:0)
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # TX rejected (too early): 0.00% (req:1, rej:0)
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # BEACON queued: 0
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # BEACON sent so far: 0
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: # BEACON rejected: 0
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: ##### END #####
Nov 24 12:59:26 klk-wifc-0300FE local1.notice spf: JSON up: {"stat":{"time":"2017-11-24 12:59:26 GMT","rxnb":1,"rxok":1,"rxfw":1,"ackr":0.0,"dwnb":1,"txnb":1}}
Nov 24 12:59:34 klk-wifc-0300FE local1.notice spf: INFO: [down] PULL_ACK received in 52 ms
Nov 24 12:59:44 klk-wifc-0300FE local1.notice spf: INFO: [down] PULL_ACK received in 52 ms
Nov 24 12:59:54 klk-wifc-0300FE local1.notice spf: INFO: [down] PULL_ACK received in 52 ms
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: ##### 2017-11-24 12:59:56 GMT #####
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: ### [UPSTREAM] ###
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # RF packets received by concentrator: 0
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # RF packets forwarded: 0 (0 bytes)
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # PUSH_DATA datagrams sent: 1 (111 bytes)
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # PUSH_DATA acknowledged: 0.00%
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: ### [DOWNSTREAM] ###
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # PULL_DATA sent: 3 (100.00% acknowledged)
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # PULL_RESP(onse) datagrams received: 0 (0 bytes)
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # RF packets sent to concentrator: 0 (0 bytes)
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # TX errors: 0
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # TX rejected (collision packet): 0.00% (req:1, rej:0)
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # TX rejected (collision beacon): 0.00% (req:1, rej:0)
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # TX rejected (too late): 0.00% (req:1, rej:0)
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # TX rejected (too early): 0.00% (req:1, rej:0)
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # BEACON queued: 0
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # BEACON sent so far: 0
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: # BEACON rejected: 0
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: ##### END #####
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: JSON up: {"stat":{"time":"2017-11-24 12:59:56 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0}}
Nov 24 12:59:56 klk-wifc-0300FE local1.notice spf: INFO: host/sx1301 time offset=(1511525188s:154028µs) - drift=12149µs

```
where I have a 52ms of response time from server.

In debug.log I got this:
```
2017-11-24 12:53:56.109 [debug] <0.292.0>@lorawan_mac:process_status:37 Gateway 7276FF00390300FE had 1 uplink CRC errors
2017-11-24 12:53:56.109 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:54:26.110 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:54:56.115 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:55:26.117 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:55:56.120 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:56:26.129 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:56:56.126 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:57:26.130 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:57:56.133 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:58:26.141 [debug] <0.292.0>@lorawan_mac:process_status:37 Gateway 7276FF00390300FE had 1 uplink CRC errors
2017-11-24 12:58:26.141 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:58:56.140 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:59:26.142 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 12:59:56.132 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 13:00:26.124 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 13:00:56.108 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 13:01:26.094 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 13:01:56.201 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 13:02:26.074 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 13:02:56.082 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 13:03:26.082 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
2017-11-24 13:03:56.081 [warning] <0.292.0> gateway 7276FF00390300FE {ack_lost,100.0}
```
Which ack is lost? Why CRC error in uplink if I got correct data?
[image]

webfrank, 2017-11-24:
Forgot to mention server is latest docker version 0.4.13

gotthardp, 2017-11-24:
The ack_lost is caused by the `"ackr":0.0` value sent by your packet_forwarder. It should indicate how may frames were acknowledged, but since no frames were sent it is a bit unclear if 100% of zero frames of 0% of zero frames were acknowledged. I consider this is a bug in the server. Will fix that in next release.

The CRC error is **not** related to any received frame. It means the gateway received some frame it couldn't decode. It is very often some noise or some very distant device. Nothing to worry about. Perhaps I should decrease severity of that log.

gotthardp, 2018-01-05:
Fixed in 0.5.0 (to be released).

closed 2018-01-05 by gotthardp

p-nahrgang, 2018-02-21:
Hi,
I have used the Version 0.4.13 and got "ack_lost" and the payload was delayed on the websocket to Node Red.
Now I have upgraded the lorawan-serever to 0.5.0 and have still "ack_lost". 
I'm using the Laird RM186 in the Node and the Laird RG1xx Gateway.
In Addition with Version 0.5.0 I got the warning - Repeated_reset

gotthardp, 2018-02-21:
Please open a separate issue and provide a gateway and server logs. Thanks.