gotthardp/lorawan-server #246: trigger event

gnob417, 2017-12-01:
Hi, 
How could I trigger periodic events on the server side like periodic downlink events?
is there any module (like timer) in erlang to trigger events on every several minutes?

regards.

gotthardp, 2017-12-01:
There is erlang timer module. Are you writing an internal application for class C or what is your use-case?

gnob417, 2017-12-05:
I want to trigger periodic downlinks for giving more information to nodes.
Could you tell me the part where downlink packets are made? (including mtype, payload..)

altishchenko, 2017-12-05:
Hello @gnob417 , to be honest with you - the best way to do it is via an external application + connector. Like websocket, mqtt or http/rest. It is a bit tricky or even not at all possible to get this type of functionality from the built-in 'plugin' structure, which you are referencing.
encode_tx is in a lorawan_handler.erl - it is a function, that goes through the list of plugins and invokes the one assigned to your mote/device via consecutive calls to invoke_handler, invoke_handler2.
The only magic there is to get the app you assigned to your device, and drive it through the standard set of callbacks (encode_tx in your case). Have a look at the code.

altishchenko, 2017-12-05:
Sorry, I sounded a bit obscure, `encode_tx` is also a callback function name you should provide in your plugin module/app.

gnob417, 2017-12-06:
so, where is the part that downlink packets are made? (including mtype, payload..)
maybe i could do like that with timer module.

and, i'm quite confused about lots of handle_cast function definition..
are they all different functions?

altishchenko, 2017-12-06:
Each module in Erlang has its own namespace, so function foo defined in one module is absolutely different from the foo in the other, hence you allways call Mod:foo to differentiate, unless this foo is defined in the module you call it from.
Erlang uses 'pattern matching' approach to dealing with functions and their arguments. This means that you can have *one* function split over several blocks (mind the ; and final .):
foo(1) -> do something once;
foo(2) -> do something twice;
foo(10) -> as much as I can;
foo(N) -> complain - don't know how to do N.
These blocks are called clauses and every clause is tried consecutively, before the matching one is found and gives the result.
Also, `handle_cast` is part of the API in OTP, it is a required 'callback' for behaviours like `gen_server`, and is used to convey asynchronous messages to the running process in the form handle_cast(Message, State).

As for the downlinks, `encode_tx` is a special function used to call the assigned application to re-encode incoming data to be sent to the device into standard structure expected by lorawan_mac process. It **must not** be used by any app directly. It does **not** generate new data on its own. It is only invoked by lorawan_mac, when it cannot recognize the data it is about to send.

For your task - there are two things you can do:
1. You can use lorawan_handler:store_frame function from your running timer, providing it with devaddr and TxData record. Which is a hack and I don't recommend it.
2. Consider driving your devices from an _external_ application. Ex., consider websocket. You set the app of your device to websocket, then you can connect to yourserver:port/ws/nodes/DevAddr/raw or json and send any data you like any time you like. This way you can bomb your nodes from a trivial python script the way you like and receive replies from them.

P.S.: If you'd like to know more about Erlang modules, functions and its structure, I strongly recommend [This book](http://learnyousomeerlang.com).
P.P.S: Bear in mind, the server is not a single process. It is a bunch of communicating processes. To get the feel of how it looks and if you are running the server on a machine with graphical desktop, do this:
1. Start your server.
2. In another terminal window run `erl -sname admin`:
```
1> net:ping(lorawan@yourcomputername).
pong
2> nodes().
[lorawan@yourcomputername]
3> observer:start().
```
When the graphical window pops-up, look for the 'Nodes' menu and choose `lorawan`. Then in the Applications tab, you will be able to explore the server process structure.

gnob417, 2017-12-07:
Thanks for kind support.
However, when i follow your suggestion about exploring server structure, error below comes up. 

> =ERROR REPORT==== 7-Dec-2017::14:03:38 ===
> WX Failed loading "wxe_driver"@"/usr/lib/erlang/lib/wx-1.8.2/priv"
> {error,{{load_driver,"symbol _ZThn704_N17wxGenericListCtrl31GetSizeAvailableForScrollTargetERK6wxSize, version WXU_3.0 not defined in file libwx_gtk2u_core-3.0.so.0 with link time reference"},
>            [{wxe_server,start,1,[{file,"wxe_server.erl"},{line,65}]},
>                     {wx,new,1,[{file,"wx.erl"},{line,115}]},
>                              {observer_wx,init,1,[{file,"observer_wx.erl"},{line,98}]},
>                                       {wx_object,init_it,6,[{file,"wx_object.erl"},{line,372}]},
>                                                {proc_lib,init_p_do_apply,3,
>                                                                      [{file,"proc_lib.erl"},{line,247}]}]}}

How could i overcome this error?

Regards.

altishchenko, 2017-12-11:
For the first problem: what operating system are you running? It looks like the installed version of WXU support is not the required 3.0 one or you just missing libwx on your machine. Can you verify, that wx support is installed in your packages.

For the second: This is a bit of black magic for me either and I hope Petr (@gotthardp ) will clarify this one. The rule I think, is - RX2 window is used, when the server detects retransmission from the device, i.e. device in its next TX does not acknowledge the receipt of a previous frame from the server.

gotthardp, 2017-12-11:
User can explicitly say which window to use for what device. If set to Auto, then
 * The Join RX window is RX1 for the odd join retransmission (1,3,5...) and RX2 for even join retransmission (2, 4, 6).
 * The Downlink RX window is RX1 if the frame processing was shorter than 800 ms and RX2 otherwise.

gotthardp, 2018-01-05:
The new 0.5.0 allows you to create your own supervision tree, start own gen_server and do whatever you want. The simplest demoapp is here: https://github.com/gotthardp/lorawan-server-demoapp

closed 2018-01-05 by gotthardp

gnob417, 2018-01-16:
@gotthardp where is the code about your comment? i could not find it.

> The Downlink RX window is RX1 if the frame processing was shorter than 800 ms and RX2 otherwise.

gotthardp, 2018-01-16:
transmit_and_delete does class A downlink of a queued frame and deletes the frame from that queue

gnob417, 2018-01-16:
@gotthardp 
hi, What i want to know is that `transmit_and_delete ` function triggers which function.
The end of `transmit_and_delete ` function, there is
`{send, TxData#txdata{port=OutPort, pending=OutPending}}`

when the function is finished, what function would be triggered through above code?

gotthardp, 2018-01-16:
Then it returns to https://github.com/gotthardp/lorawan-server/blob/master/src/lorawan_handler.erl#L154 and the send_unicast is called

gnob417, 2018-01-17:
do you mean `send_unicast` is called directly rather than calling `uplink()` which has `send_unicast` function?

gotthardp, 2018-01-17:
Uplink is the one who called the handler that called the transmit-and-delete. When transmitanddelete returns, it returns to the uplink function.