gotthardp/lorawan-server #249: clean way to run lorawan_server build as startup service

kazi93, 2017-12-06:
@gotthardp @altishchenko 
What is the clean way to run lorawan_server build as startup service that is :
lorawan-server/_build/default/rel/lorawan-server/bin$ sudo ./lorawan-server

altishchenko, 2017-12-06:
@kazi93 for Ubuntu you can generate a .deb package with `make dpkg` and install it as any other usual package. For other situations, where you don't want a package, you can use `rebar3 tar`, which will make a `tar.gz` bundle required to run it. lorawan-server uses, what is called, a 'simple startup' approach and that leaves no other options :) Look at the scripts in the `scripts` directory of the repository.

altishchenko, 2017-12-06:
And, as a follow up, you are definitely not going to run it from `_build/default/rel/...` etc.

kazi93, 2017-12-07:
@altishchenko Thanks. Ubuntu it is perfect.

closed 2017-12-07 by kazi93