gotthardp/lorawan-server #251: Suggestion: REST "Web hooks" for Device OTAA join/unjoin

Eric24, 2017-12-11:
From a database standpoint (assuming an OTAA operational mode), Devices are the objects that there will be most of in any given application. While it's convenient to store Device configurations within the Mnesia database, it would be a nice alternative to allow those configurations to be stored externally, and have the server request authentication and configuration data from that external source upon receiving a 'join'.

When a device attempts to join, instead of looking up the device in the local Mnesia database, make a "join" REST call to the configured third-party server. The server will authenticate the device and return its configuration data (as JSON). Once the device is authenticated, it is stored in the Mnesia database as a Node until it is normally removed (i.e. communications timeout). And at that time, a separate "unjoin" REST transaction is made to the same third-party server, simply providing a notification that the device is no longer active.

As an added enhancement, multiple third-party servers could be configured (similar to Connectors), possibly based on a partial match, RegEx, or range-check of the DevEUI.

gotthardp, 2017-12-11:
This sound like a support for an external Join Server. Is there some existing Join Server we could integrate with?

altishchenko, 2017-12-11:
Or, at least try to?

Alexander Tishchenko
M: +7 (916) 704-61-57
E: atishch@gmail.com

Eric24, 2017-12-11:
Kind of the point is to make this interface as generic as possible, so as to make it easy for someone to "be" a join server. If there's a good/generic interface already in-use, that might certainly be an option; or at least to consider it as a model.

But what's really needed for this interface? The function of the join server is to authenticate a device that wants to join, and then to return some relatively-static configuration information, that persists on the Network Server until the device unjoins or times out. It's fairly simple and doesn't really have much to do with the functionality of LoRaWAN, other than the LoRaWAN-specific configuration data (which could be fairly opaque to the join server). Am I missing something?

Eric24, 2017-12-11:
By the way, if there's interest in developing this interface, I'll volunteer to write the test join server.

gotthardp, 2017-12-11:
Frankly speaking, if there is no real Join Server to integrate with, why shall we implement the interface?

Eric24, 2017-12-11:
So it's possible to _implement_ a join server, using the lorawan server to manage everything but the device authentication and configuration storage. Currently, device configuration data must be stored within the lorawan server's mnesia database (and thus join authentication must be handled locally). While the device database can be easily managed through the API, I'm just suggesting that the option be provided to allow device data to be stored and join authentication be handled elsewhere. I suggested a REST interface so that such a service can be easily implemented in virtually any language (and I would expect that it would be rarely implemented as a stand-alone "join server", but rather a "join service" as part of a larger system).

gotthardp, 2017-12-11:
I see.

altishchenko, 2017-12-12:
Sorry for interruption, but what prevents someone implementing "Configuration management server" on another node making a mnesia cluster? IMHO, this way you can have multiple servers running on different nodes and yet share the same configuration data? Implementing the REST feature will require a lot of internal hacking (as of yet), implementing standalone "Config server" - just implementing the server to works with underlying mnesia.

gotthardp, 2017-12-12:
Of course you can do that, but I understood the need(?) was to have the Config server as a non-Erlang component.

altishchenko, 2017-12-12:
Well, possibly yes. But I like the mnesia idea, i'll try it out as soon as I get into the lab. Btw, as a by-product any changes made on any of the servers directly will be visible from other servers and config server as well.

gotthardp, 2017-12-12:
To be honest I see much higher user demand for high-availability clustering rather than for clusters isolating certain functions. Any experiments with high-availability clusters would be highly appreciated.

Eric24, 2017-12-12:
Clustering for high-availability? Yes, please!

But yes, the idea for the REST "hooks" for device config storage and authentication was to allow devices to be managed by a non-Erlang system. I love Erlang for this application, but there are many other (more common) languages/environments that are used for "upstream" application development.

That said, if the lorawan server could be clustered in HA mode, I would be much more comfortable allowing its mnesia database to be the "authoritative" source of the device config data, managed by the "upstream" application via the admin API.

gotthardp, 2017-12-12:
Yeah, having lorawan as HA cluster, with mnesia as the authoritative source for config data is my aim.

altishchenko, 2017-12-14:
Working on it. Re-reading these chapters from DSE.

gotthardp, 2017-12-14:
@altishchenko That is awesome. Now it's the best time to modify the architecture (if necessary).

altishchenko, 2017-12-14:
@gotthardp Do you have a 5.0 timeline in place, so I can plan accordingly? Or a rough estimate for a release date?

gotthardp, 2017-12-14:
@altishchenko Merge of dev to master ~mid January, release at lest a month later.

altishchenko, 2017-12-14:
Petr (@gotthardp ) so we don't have a lot of time. There are a few points I'd like you to consider, before moving ahead. I'll summarize them in an e-mail and that will concern abstracting the db/config layer as well. As a pre-hint, I'd strongly suggest trying to follow OTP blueprints and switching over either to 'start_script' or explicit 'boot' config (a.k.a. 'bin/start'). I'll explain my points later. Just to let you know - 'boot' works, 'start_script' works as well. Even distributing erts and stdlib works (I'd say perfect - I am building on MacOS _for Ubuntu_ and I don't have to worry about Ubuntu and its packages at all...). Remote boot-strapping also works, only mnesia has to be configured carefully in this case. Currently, I am looking at your usage of syn:, you introduced into the code some time ago. I do hope it is not critical (if we are heading for HA).

altishchenko, 2017-12-14:
> That said, if the lorawan server could be clustered in HA mode, I would be much more comfortable allowing its mnesia database to be the "authoritative" source of the device config data, managed by the "upstream" application via the admin API.

Can I say that this rounds it up a bit? I am not feeling comfortable with two configs staying in two different places.

Let me recite:

1. OTAA devices should be able to be accepted and configured from some central source (via GET, possibly).
2. Given the current server - the process of assignment of the DevAddr and the keys (NwsK, AppSkey) is kept strictly to the server instance, so the others will not know about them (mnesia is out?).
3. Hence, there should be another backwards interface to *POST* the selected and assigned values. No?

There are also questions on behalf of received frames, keep-alives, etc, I won't mention them here.

Also, there is another question: ABP devices. What is going to happen to them? They are not OTAAs - they should just register. But if they are just plain ABPs? Should we post their config over to the central server as well? Or, where should we enter this config at all? Any consistency here?

It might be a lot easier, if you will explain to us your business model/logic. No, really! There are options and a lot of them! Especially considering things to be `Erlang` or `Python/Java/C++/etc` which can natively communicate with `Erlang` anyway, or just do the tricks and never bother.

Petr (@gotthardp )? 

FWIW: I don't like the idea. It is pulling the overall 'general subroutine blanket' over the eyes and trying to build a universal coffee machine, that'll brew anything from LUA to Assembler. (Sorry, if I insulted anyone, but from my experience, jumping into this sort of proposals always ends in wrecks).

gotthardp, 2017-12-15:
@altishchenko let's do plain HA first, i don't want to split the database either.

Eric24, 2017-12-15:
To be clear, I brought up this idea because I was uncomfortable with non-HA mnesia as the authoritative database for devices. With HA, I'm a lot more comfortable with this, and can easily see letting 100% of the config data be stored in mnesia, with just remote management from the "upstream" application via the existing API.

altishchenko, 2017-12-15:
Ok, I'll focus on HA.

altishchenko, 2017-12-25:
@Eric24, can you drop me an e-mail at atishch@gmail.com, so we can discuss your HA expectations?