gotthardp/lorawan-server #252: Microchip-Loramote OTAA/ABP

kazi93, 2017-12-12:
@gotthardp 
Anyone configured and connect with Loramote RN2483? I do not see and node creation in the server. Help required. I am using LoRa Development Utility as below attached.
[image]

01:54:26.975 > mac set deveui 0004A30B001B26BF
01:54:26.986 > ok
01:54:26.987 > mac set appeui 1000000000000000
01:54:26.997 > ok
01:54:26.998 > mac set appkey 2B7E151628AED2A6ABF7158809CF4F3C
01:54:27.013 > ok
01:54:27.013 > mac save
01:54:28.148 > ok
01:56:38.643 > mac join otaa
01:56:38.672 > ok

gotthardp, 2017-12-12:
Hello. Standard question: what is in the server debug.log?

altishchenko, 2017-12-12:
And server side device config screenshot, please.

kazi93, 2017-12-22:
@altishchenko I switched back to ABP for primary test. However, now the error is : 

warning | 2017-12-22 06:21:53 | 2017-12-22 06:21:53 | 1 | server |   | http_error | {404,"/devstat/26011B64"}
-- | -- | -- | -- | -- | -- | -- | --
  | warning | 2017-12-22 06:21:48 | 2017-12-22 06:21:48 | 1 | server |   | http_error | {404,"/devstat/26011B64"}

debug.log

[image]

gotthardp, 2017-12-22:
This warning is not an error. It is shown also in a situation when the device didn't send any frames yet, which is your case (FCnt=0). Once at least one frame is received the warnings will stop.

gotthardp, 2017-12-28:
the http_error warnings will no longer be shown in the future 0.5.0

closed 2017-12-28 by gotthardp