gotthardp/lorawan-server #254: dbimport Connection refused

hallard, 2017-12-13:
Hi,
I've done a dbexport from one server (PI+ Ic880A) and do the dbimport to another one (Pi Zero + RAK831) and on import I got some error messages
```
/usr/lib/lorawan-server/bin $ sudo ./dbimport backup-20171207-140857/
failed: Connection refused.
Authentication selected: Basic realm="lorawan-server"
2017-12-13 12:30:45 URL:http://localhost:8080/connectors [0] -> "connectors" [1]
failed: Connection refused.
Authentication selected: Basic realm="lorawan-server"
2017-12-13 12:30:45 URL:http://localhost:8080/devices [0] -> "devices" [1]
failed: Connection refused.
Authentication selected: Basic realm="lorawan-server"
2017-12-13 12:30:45 URL:http://localhost:8080/gateways [0] -> "gateways" [1]
failed: Connection refused.
Authentication selected: Basic realm="lorawan-server"
2017-12-13 12:30:46 URL:http://localhost:8080/handlers [0] -> "handlers" [1]
failed: Connection refused.
Authentication selected: Basic realm="lorawan-server"
2017-12-13 12:30:46 URL:http://localhost:8080/ignored_nodes [0] -> "ignored_nodes" [1]
failed: Connection refused.
Authentication selected: Basic realm="lorawan-server"
2017-12-13 12:30:46 URL:http://localhost:8080/multicast_channels [0] -> "multicast_channels" [1]
failed: Connection refused.
Authentication selected: Basic realm="lorawan-server"
2017-12-13 12:30:46 URL:http://localhost:8080/nodes [0] -> "nodes" [1]
failed: Connection refused.
Authentication selected: Basic realm="lorawan-server"
2017-12-13 12:30:46 URL:http://localhost:8080/users [0] -> "users" [1]
```

Everything looks imported (gateway, nodes, connectors, ...) 
but I'm wondering why I've got these error messages?
When I done this one same server (to restore) I didn't had any errors of course both server have same default credentials admin/admin.

gotthardp, 2017-12-13:
Hmmm. I saw similar messages and I thought it's caused by the HTTP authentication procedure (where the first request is rejected with a request for credentials). However, I agree the message is confusing.
I will keep this issue open and try to avoid this issue.

hallard, 2017-12-13:
Makes sense, I tried at first time to pass credential on command line but did not found how to and even if it's possible.

altishchenko, 2017-12-14:
This is unfortunate curl 'feature'.

closed 2017-12-28 by gotthardp