gotthardp/lorawan-server #255: starting with init.d does not work

hallard, 2017-12-14:
I'm trying to run the server on a LorixOne (yocto), I've made an SD image that I put into the Lorix One gateway and all is fine (network, packet forwarder, lorawan-server, ..) but I'm able to manually start the lorawan-server service

As you can see all is in /usr/lib/lorawan-server is owned my lorawan user so all looks good
```
ttn-gw06:/usr/lib$ pwd
/usr/lib
ttn-gw06:/usr/lib$ ll lorawan-server/
total 316
drwxr-xr-x  7 lorawan lorawan   1024 Dec 14 10:47 ./
drwxr-xr-x 19 root    root      5120 Oct 24 00:24 ../
drwxr-xr-x  2 lorawan lorawan   1024 Dec 14 10:36 'Mnesia.lorawan@ttn-gw06'/
drwxr-xr-x  2 lorawan lorawan   1024 Oct 13 13:04 bin/
-rw-r-----  1 lorawan lorawan 311893 Dec 14 10:47 erl_crash.dump
drwxr-xr-x 16 lorawan lorawan   1024 Oct 13 13:04 lib/
drwxr-xr-x  2 lorawan lorawan   1024 Dec 14 10:36 log/
drwxr-xr-x  3 lorawan lorawan   1024 Oct 13 13:04 releases/
```

running as lorawan user works fine I can go to web interface on 8080
```shell
ttn-gw06:/usr/lib/lorawan-server$ sudo su lorawan
ttn-gw06:/usr/lib/lorawan-server$ bin/lorawan-server
```

running as admin (default lorix user) with sudo works fine also I can go to web interface on 8080
```shell
ttn-gw06:/usr/lib/lorawan-server$ whoami
admin
ttn-gw06:/usr/lib/lorawan-server$ sudo bin/lorawan-server`
```

but trying to auto start the service does not work, I even tried to put all on the same folder (log)
`/usr/lib/loarawan-server` to see but nothing, it start then crash

here my `/etc/init.d/lorawan-server`

```shell
#!/bin/bash

NAME="lorawan-server"
ENABLED="yes"

[ -f /etc/default/$NAME ] && source /etc/default/$NAME

run_dir=/var/run/lora
server_pidfile=$run_dir/lorawan-server-1.pid

ask_sudo()
{
    sudo -nv 2> /dev/null
    SUDOCREDCACHED=$?
    if [ $SUDOCREDCACHED != 0 ] ; then
            # acquire credentials
        sudo -v
        if [ $? != 0 ] ; then
            exit 1
        fi
    fi
}

do_start() {
    echo -n "Starting $NAME: "

     mkdir -p $run_dir
    /sbin/start-stop-daemon --background --start --chuid lorawan:lorawan \
        --make-pidfile --pidfile $server_pidfile --startas /bin/bash -- \
        -c "LORAWAN_HOME=/usr/lib/lorawan-server LORAWAN_LOG_ROOT=/usr/lib/lorawan-server/log /usr/lib/lorawan-server/bin/lorawan-server"

    echo "OK"
}

do_stop() {
    echo -n "Stopping $NAME: "
    su -c "/usr/lib/lorawan-server/bin/lorawanctl stop" lorawan
    rm -f $server_pidfile
    echo "OK"
}

if [ "$ENABLED" != "yes" ]; then
    echo "$NAME: disabled in /etc/default"
    exit
fi

case "$1" in
    "start")
    ask_sudo
    do_start
        ;;
    "stop")
    ask_sudo
    do_stop
        ;;
    "restart")
        ## Stop the service and regardless of whether it was
        ## running or not, start it again.
        ask_sudo
        do_stop
        do_start
        ;;
    *)
        ## If no parameters are given, print which are avaiable.
        echo "Usage: $0 {start|stop|restart}"
        exit 1
    ;;
esac
```

I'm pretty sure it's permission related but I have no trace, no log of the error so I'm stuck, so how can I see log ? Any idea of the issue ?

Thanks

gotthardp, 2017-12-14:
Tough problem. Did you try running the `start-stop-daemon` command from the console to see if that works? Some more error can be reported from the `start-stop-daemon` in syslog. Is there something?

hallard, 2017-12-14:
yeah I tried that but no output

```
ttn-gw06:/usr/lib/lorawan-server$ sudo /sbin/start-stop-daemon --background --start --chuid lorawan:lorawan --make-pidfile --pidfile /var/run/lora/lorawan-server-1.pid --startas /bin/bash -- -c "LORAWAN_HOME=/usr/lib/lorawan-server LORAWAN_LOG_ROOT=/usr/lib/lorawan-server/log /usr/lib/lorawan-server/bin/lorawan-server"
ttn-gw06:/usr/lib/lorawan-server$
```

then you give an idea, remove the --background and here it is

```
ttn-gw06:/usr/lib/lorawan-server$ sudo /sbin/start-stop-daemon --start --chuid lorawan:lorawan --make-pidfile --pidfile /var/run/lora/lorawan-server-1.pid --startas /bin/bash -- -c "LORAWAN_HOME=/usr/lib/lorawan-server LORAWAN_LOG_ROOT=/usr/lib/lorawan-server/log /usr/lib/lorawan-server/bin/lorawan-server"
2017-12-14 11:24:43 {{error,"Error when reading /home/root/.erlang.cookie: eacces"},[]}
2017-12-14 11:24:43 crash_report
    initial_call: {auth,init,['Argument__1']}
    pid: <0.47.0>
    registered_name: []
    error_info: {error,{error,"Error when reading /home/root/.erlang.cookie: eacces"},[{auth,init_cookie,0,[{file,"auth.erl"},{line,286}]},{auth,init,1,[{file,"auth.erl"},{line,140}]},{gen_server,init_it,2,[{file,"gen_server.erl"},{line,365}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,333}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}
    ancestors: [net_sup,kernel_sup,<0.36.0>]
    message_queue_len: 0
    messages: []
    links: [<0.45.0>]
    dictionary: []
    trap_exit: true
    status: running
    heap_size: 610
    stack_size: 27
    reductions: 560
2017-12-14 11:24:43 supervisor_report
    supervisor: {local,net_sup}
    errorContext: start_error
    reason: {{error,"Error when reading /home/root/.erlang.cookie: eacces"},[{auth,init_cookie,0,[{file,"auth.erl"},{line,286}]},{auth,init,1,[{file,"auth.erl"},{line,140}]},{gen_server,init_it,2,[{file,"gen_server.erl"},{line,365}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,333}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}
    offender: [{pid,undefined},{id,auth},{mfargs,{auth,start_link,[]}},{restart_type,permanent},{shutdown,2000},{child_type,worker}]
2017-12-14 11:24:43 supervisor_report
    supervisor: {local,kernel_sup}
    errorContext: start_error
    reason: {shutdown,{failed_to_start_child,auth,{{error,"Error when reading /home/root/.erlang.cookie: eacces"},[{auth,init_cookie,0,[{file,"auth.erl"},{line,286}]},{auth,init,1,[{file,"auth.erl"},{line,140}]},{gen_server,init_it,2,[{file,"gen_server.erl"},{line,365}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,333}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}}}
    offender: [{pid,undefined},{id,net_sup},{mfargs,{erl_distribution,start_link,[]}},{restart_type,permanent},{shutdown,infinity},{child_type,supervisor}]
2017-12-14 11:24:43 crash_report
    initial_call: {application_master,init,['Argument__1','Argument__2','Argument__3','Argument__4']}
    pid: <0.35.0>
    registered_name: []
    error_info: {exit,{{shutdown,{failed_to_start_child,net_sup,{shutdown,{failed_to_start_child,auth,{{error,"Error when reading /home/root/.erlang.cookie: eacces"},[{auth,init_cookie,0,[{file,"auth.erl"},{line,286}]},{auth,init,1,[{file,"auth.erl"},{line,140}]},{gen_server,init_it,2,[{file,"gen_server.erl"},{line,365}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,333}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}}}}},{kernel,start,[normal,[]]}},[{application_master,init,4,[{file,"application_master.erl"},{line,134}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}
    ancestors: [<0.34.0>]
    message_queue_len: 1
    messages: [{'EXIT',<0.36.0>,normal}]
    links: [<0.34.0>,<0.33.0>]
    dictionary: []
    trap_exit: true
    status: running
    heap_size: 987
    stack_size: 27
    reductions: 248
2017-12-14 11:24:43 std_info
    application: kernel
    exited: {{shutdown,{failed_to_start_child,net_sup,{shutdown,{failed_to_start_child,auth,{{error,"Error when reading /home/root/.erlang.cookie: eacces"},[{auth,init_cookie,0,[{file,"auth.erl"},{line,286}]},{auth,init,1,[{file,"auth.erl"},{line,140}]},{gen_server,init_it,2,[{file,"gen_server.erl"},{line,365}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,333}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}}}}},{kernel,start,[normal,[]]}}
    type: permanent
{"Kernel pid terminated",application_controller,"{application_start_failure,kernel,{{shutdown,{failed_to_start_child,net_sup,{shutdown,{failed_to_start_child,auth,{{error,\"Error when reading /home/root/.erlang.cookie: eacces\"},[{auth,init_cookie,0,[{file,\"auth.erl\"},{line,286}]},{auth,init,1,[{file,\"auth.erl\"},{line,140}]},{gen_server,init_it,2,[{file,\"gen_server.erl\"},{line,365}]},{gen_server,init_it,6,[{file,\"gen_server.erl\"},{line,333}]},{proc_lib,init_p_do_apply,3,[{file,\"proc_lib.erl\"},{line,247}]}]}}}}},{kernel,start,[normal,[]]}}}"}
Kernel pid terminated (application_controller) ({application_start_failure,kernel,{{shutdown,{failed_to_start_child,net_sup,{shutdown,{failed_to_start_child,auth,{{error,"Error when reading /home/root

Crash dump is being written to: erl_crash.dump...done
ttn-gw06:/usr/lib/lorawan-server$
```

gotthardp, 2017-12-14:
The question is why is he reading the cookie from `/home/root`? It should be read from whatever the home for the `lorawan` user is, most preferrably the `/usr/lib/lorawan-server`.

hallard, 2017-12-14:
I think it's because of sudo, but I'm not sure, and as init.d script are launched by root that may be the issue. But it's very strange, on another target with systemd it works fine.

hallard, 2017-12-14:
Solved !!

Change in `/etc/init.d/lorawan-server`  

```
do_start() {
    echo -n "Starting $NAME: "

     mkdir -p $run_dir
    /sbin/start-stop-daemon --background --start --chuid lorawan \
        --make-pidfile --pidfile $server_pidfile --startas /bin/bash -- \
        -c "LORAWAN_HOME=/usr/lib/lorawan-server LORAWAN_LOG_ROOT=/usr/lib/lorawan-server/log /usr/lib/lorawan-server/bin/lorawan-server"

    echo "OK"
}
```
by 
```
do_start() {
    echo -n "Starting $NAME: "

     mkdir -p $run_dir
    /sbin/start-stop-daemon --background --start -u lorawan \
        --make-pidfile --pidfile $server_pidfile --startas /bin/bash -- \
        -c "LORAWAN_HOME=/usr/lib/lorawan-server LORAWAN_LOG_ROOT=/usr/lib/lorawan-server/log /usr/lib/lorawan-server/bin/lorawan-server"

    echo "OK"
}
```
in fact this `--chuid lorawan` should be `-u lorawan`

closed 2018-01-24 by gotthardp