gotthardp/lorawan-server #256: Adding custom web page

hallard, 2017-12-14:
For a customer I need to have a basic web page (html + css + js) that connect with websocket to lorawan-server but without any authentication. All need to work on same device

At first I installed nginx and all works fine but I would like to avoid this and use the lorawan-server webserver. So I created a `default.html` page in root web `/usr/lib/lorawan-server/lib/lorawan_server-0.4.12/priv`

```
ttn-gw06:/usr/lib/lorawan-server/lib/lorawan_server-0.4.12/priv$ ll
total 8
drwxr-xr-x 4 lorawan lorawan 1024 Dec 14 11:14 ./
drwxr-xr-x 5 lorawan lorawan 1024 Oct 13 13:04 ../
drwxr-xr-x 3 lorawan lorawan 1024 Oct 13 13:04 admin/
-rw-r--r-- 1 lorawan lorawan  612 Dec 14 11:13 default.html
-rw-r--r-- 1 lorawan lorawan 1078 Oct 13 13:03 favicon.ico
-rw-r--r-- 1 lorawan lorawan   99 Oct 13 13:03 root.html
```
but trying `http://192.168.1.50:8080/default.html` fire a connection refused.

So I I also tested with a subfolder `custom` and put the html page into
```
ttn-gw06:/usr/lib/lorawan-server/lib/lorawan_server-0.4.12/priv$ ll
total 8
drwxr-xr-x 4 lorawan lorawan 1024 Dec 14 11:14 ./
drwxr-xr-x 5 lorawan lorawan 1024 Oct 13 13:04 ../
drwxr-xr-x 3 lorawan lorawan 1024 Oct 13 13:04 admin/
drwxr-xr-x 2 lorawan lorawan 1024 Dec 14 11:14 custom/
-rw-r--r-- 1 lorawan lorawan 1078 Oct 13 13:03 favicon.ico
-rw-r--r-- 1 lorawan lorawan   99 Oct 13 13:03 root.html
```
But again, trying `http://192.168.1.50:8080/custom/default.html` fire a connection refused.

I think I need to add the folder or file somewhere in the cowboy configuration?

Thanks

gotthardp, 2017-12-14:
The allowed pages are in `src/lorawan_app.erl`. Adding a file under `admin` shall work without any change... and it doesn't need authentication :) It's a security flaw though.

hallard, 2017-12-14:
Got it thanks, but I need to recompile my SD image, correct ?

gotthardp, 2017-12-14:
No, why? I think you can just put the file into `/usr/lib/lorawan-server/lib/lorawan_server-0.4.12/priv/admin`.

hallard, 2017-12-14:
Correct, Sorry, I was not talking about html file but the config file `src/lorawan_app.erl` I need to change ;-)

gotthardp, 2017-12-14:
Everything under /admin is served by default. You need to change it only if you want to serve it from some other directory.

hallard, 2017-12-14:
Confirmed putting file in admin works without any authentication, thanks.
I'll check later to put custom pages in another folder to stay clean ;-)

closed 2017-12-14 by hallard