gotthardp/lorawan-server #258: Single-Channel "Nano" Gateway Support

Eric24, 2017-12-19:
Does anyone have experience using a single-channel gateway (like pycom.io, etc.) with this server? Of course it's not "true" LoRaWAN (since it can only listen on one channel at a time), but such gateways are compatible with TTN, for example.

Does this server currently support single-channel gateways? If so, what's required to set it up? If not, what limitations prevent it? Also, what needs to be considered for such a gateway (other than the obvious limit to the number of supported devices)?

gotthardp, 2017-12-19:
People tried. You basically need to convince **both** the device and the server to use this single channel for downlink. On the server side this can be done by switching the "TX Window" for all connected devices to TX2.

Eric24, 2017-12-19:
Interesting. So exactly how does using TX2 make this work?

javierpedrido, 2017-12-19:
Hi,

I have a Pycom Lopy acting as a gateway of some Multitech's mDots and xDots and this server works fine. One other thing to consider is that, some end node's firmware do not allow to modify the OTA join messages data rate, that means that only ABP can be used for that nodes.

Regards,
Javier

Eric24, 2017-12-20:
@javierpedrido : Can you expand on what the issue is with using OTA?

javierpedrido, 2017-12-20:
Since the gateway is listening to one frequency and data rate and the Ota join message data rate is random, the gateway won't be able to pick it up. The random data rate logic cannot be changed in some node firmwares. When you have a normal gateway it can listen at the same time different frequencies and data rates.

Eric24, 2017-12-20:
@javierpedrido : Ah, that makes sense (the nodes we're using can be set to a fixed data rate, so that's why I hadn't considered that issue).

gotthardp, 2017-12-28:
I believe the question was answered. Feel free to reopen if more information is needed.

closed 2017-12-28 by gotthardp