gotthardp/lorawan-server #263: gateway & server

gnob417, 2018-01-03:
Hi, Is it possible that gateway receive lora packets but server can not recognize them?
my end node periodically sends packets.

gotthardp, 2018-01-03:
wierd things can happen, but the server always reports failures. please have a look in the debug.log and search for errors and warnings

gnob417, 2018-01-04:
suspicious logs are below

> 2018-01-04 12:14:01.761 [error] <0.285.0>@lorawan_worker:handle_cast:31 {undef,[{lorawan_mac_region,set_channels,[<<"KR920-923">>,{6,5,[{0,6}]},[]],[]ac_commands,build_fopts,2,[{file,"/home/server/12.13server/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,37}]}c_commands,handle,4,[{file,"/home/server/12.13server/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,31}]},{lorale_uplink,5,[{file,"/home/server/12.13server/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,451}]},{lorawan_worker,handlile,"/home/server/12.13server/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_worker.erl"},{line,22}]},{wpool_utils,do_try,1,[{file,"/hom13server/lorawan-server/_build/default/lib/worker_pool/src/wpool_utils.erl"},{line,59}]},{wpool_process,handle_cast,2,[{file,"/home/server/12.13serverver/_build/default/lib/worker_pool/src/wpool_process.erl"},{line,155}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,616}]}]}
> 
> 
> 
> 2018-01-04 11:30:24.678 [debug] <0.294.0>@lorawan_mac:handle_uplink:438 ADR indicator set to 1
> 2018-01-04 11:30:24.686 [debug] <0.60.0> Supervisor kernel_safe_sup started pg2:start_link() at pid <0.366.0>
> 2018-01-04 11:30:24.686 [warning] <0.285.0>@lorawan_application_websocket:handle_rx:28 Frame not sent to any process
> 2018-01-04 11:30:24.686 [warning] <0.294.0>@lorawan_application_websocket:handle_rx:28 Frame not sent to any process
> 2018-01-04 11:30:32.092 [warning] <0.285.0>@lorawan_application_websocket:handle_rx:28 Frame not sent to any process
> 2018-01-04 11:30:32.416 [debug] <0.279.0>@lorawan_mac:process_status:37 Gateway B827EBFFFEBDF7D8 had 2 uplink CRC errors
> 2018-01-04 11:30:35.495 [warning] <0.285.0>@lorawan_application_websocket:handle_rx:28 Frame not sent to any process
> 2018-01-04 11:30:39.320 [warning] <0.285.0>@lorawan_application_websocket:handle_rx:28 Frame not sent to any process
> 2018-01-04 11:30:43.743 [warning] <0.285.0>@lorawan_application_websocket:handle_rx:28 Frame not sent to any process
> 2018-01-04 11:30:48.705 [warning] <0.285.0>@lorawan_application_websocket:handle_rx:28 Frame not sent to any process
> 2018-01-04 11:30:54.906 [warning] <0.285.0>@lorawan_application_websocket:handle_rx:28 Frame not sent to any process
> 2018-01-04 11:30:57.067 [warning] <0.285.0>@lorawan_application_websocket:handle_rx:28 Frame not sent to any process
> 2018-01-04 11:31:02.420 [debug] <0.279.0>@lorawan_mac:process_status:37 Gateway B827EBFFFEBDF7D8 had 4 uplink CRC errors

Could you explain what's the problem?

gotthardp, 2018-01-04:
Thanks.
1) There is a bug in the server. I apparently never tested the korean band. I will fix it in the coming release, but for now you should set ADR to Disabled.
2) You are using a websocket backend, but there is no websocket connected to frames from this device. You may have mistyped the deveui, devaddr or group name.

gnob417, 2018-01-04:
what kind of bug about adr? 
can you tell me? maybe i can contribute something.

gotthardp, 2018-01-04:
Well, it's the 

2018-01-04 12:14:01.761 [error] <0.285.0>@lorawan_worker:handle_cast:31 {undef,[{lorawan_mac_region,set_channels,[<<"KR920-923">>,{6,5,[{0,6}]},[]]

I thought the lorawan_mac_region:set_channels/3 function is not defined for <<"KR920-923">>, but it is. There should be some other reason why Erlang thinks this function is undefined.

gotthardp, 2018-01-12:
Hello. Do you still have this issue, or may I close this?