gotthardp/lorawan-server #264: unknown_appid

Mels67, 2018-01-03:
I do get this red box all the time: unknown_appid
I also do not get any acknowledge back to my Lora device?
[image]

waynestallwood-kcom, 2018-01-03:
This caught me out too and it's due to there being no associated backend for the received data.

Have you defined some sort of connector, what are you doing with the data once it hits the lorawan-server ? 

Everything needs to be in a group of the same name (if your devices are then nodes will inherit the device group when the device first joins) Then if you have set the application as (backend) then the handler and connector also need to be in the same group as the devices/nodes.

At the moment if you look in (Received Frames) you'll probably find the application field is blank for the frames from your device.

You'll only get an ack back from the gateway if the device is sending a confirmed uplink, and be careful as too many confirmed uplinks (given the frequency you are sending packets) might cause your gateway to go over the permitted radio on time duty cycle for your area.

Mels67, 2018-01-04:
Thanks Wayne, that was the trick. I already had a mqtt connector but no handler and devices and gateway not in a group! Since I changed it, everything works fine now and MQTT server is receiving the data. thanks again!

closed 2018-01-04 by Mels67