gotthardp/lorawan-server #265: Battery value is not populated to websocket json

hosek, 2018-01-03:
Hi,
I can see battery level in Nodes section, but I cannot found it in json message received by  websocket client, this is bug or it has some reason?
Thank you in advance

gotthardp, 2018-01-03:
Hello. It is simply not implemented. I will do that for the coming release.

kazi93, 2018-01-03:
@gotthardp I guess, the same reason I cannot see data in MQTT through backend.
2018-01-03 23:32:15.724 [error] <0.308.0>@lorawan_application_backend:data_to_fields:222 Fun failed error:function_clause

Could you please what should be the modified Uplink parsing : 
fun(_Port, <<16#01, Temp:16/signed-integer, 16#07, VDD:16>>) ->
  #{field1 => Temp/10, field2 => VDD}
end.

gotthardp, 2018-01-03:
It's not the same reason.

The function is OK. I guess it does not match the frame received.

kazi93, 2018-01-04:
@gotthardp this is the data in the received frames.. could u please help..
0100DB022A070DC8

gotthardp, 2018-01-04:
Your data are 8 bytes, but the function matches only 6 bytes. There appear to be two 16-bit numbers before the 16#07 byte.

kazi93, 2018-01-04:
@gotthardp You are right. It was humidity value. I am using the previous settings now that is only temp and VDD however, same fun error on the console ...please help. 0100D90229070DB9. However, I will check with the server version 0.4.12.

gotthardp, 2018-01-04:
You need to add the Humidity to the binary then:
```erlang
fun(_Port, <<16#01, Temp:16/signed-integer, Humidity:16/signed-integer, 16#07, VDD:16>>) ->
#{field1 => Temp/10, field2 => VDD, field3 => Humidity}
end.
```

gotthardp, 2018-01-05:
The battery status will be included in 0.5.0.

closed 2018-01-05 by gotthardp