gotthardp/lorawan-server #266: adr_req_failed

Mels67, 2018-01-04:
Everything is running nice and smooth but I get all the time these yellow "adr_req_failed" on my status screen. What does it mean and how to solve? 
Thanks!
[image]

gotthardp, 2018-01-04:
Your device is rejecting the datarate requested by server's ADR algorithm. If you post here also the debug.log I can tell you if it's a server or device fault.

Mels67, 2018-01-08:
Hi Petr, seems to be solved now, no more yellow warnings. Another question. I’m sending the received data to my MQTT server (Mosquitto). Does the LoRa WAN server just post the data as received to MQTT or does it add the device address etc.... After this MQTT server I’m using Node-Red to filter and display the data but I’ve more Lora devices connected to the gateway so I first want to filter on device address.... thanks!

Mels67, 2018-01-08:
What I mean is that I’m missing things in my data like Gateway ID, rssi level, snr level and device address. This data is known by the LoraWan server but not send to my MQTT server. Can I add this data by configuration in LoraWAN server? Thanks!

gotthardp, 2018-01-08:
Hello. In the "Handler" configuration you can add "Fields". These are additional information the server adds when forwarding a frame to an external backend.

Mels67, 2018-01-08:
Hi Petr, thanks. I should buy glasses and read better, I changed in the handler RAW to JSON and now get JSON frames including all the needed info! Thanks!

closed 2018-01-08 by Mels67