gotthardp/lorawan-server #267: Class C, RXWIN2 not receive data confirmed

matiasz8, 2018-01-04:
Hello, I'm testing the class C for my semtech-mote.

I has configure the node as you can see at the picture. The node has prepare to receive through the "RXWIN1" and "RXWIN2". I has setting the Application as WebSocket and TX Window as Auto.

I have performed the following test:

I's sended packets with the json:
{"devaddr":"00020505", "data":"3531", "port":2, "time":"immediately", "confirmed":true}

What I can see is that the module receives packets as if they were in A mode, that is, once I send from the module a message asks if there are messages to download and brings them. (I's attach a picture about it)

Then, I has change TX Window to "RX2" and I's tried to receive packets with the same json, but in this case, I's not received anything.

I's tried to receive the acknowledge from a package thought the RX2 but nothing happens. 

Could you help me, please? The RX2 option, is working right?

Thanks a lot,

Nahuel

I have the release 0.4.13
I have this in the debug file:
2018-01-04 14:05:30.424 [warning] <0.315.0> node 00020505 downlink_lost
2018-01-04 14:05:30.424 [debug] <0.315.0>@lorawan_mac:handle_uplink:481 00020505 retransmitting

[image]
[image]

gotthardp, 2018-01-04:
Hello. Could you please decouple the two issues you may have found: the class C and the confirmation?

1) Class C messages are sent in RX2. No server-side configuration is requires. Just to send the message as you did and await frame in RX2. This works or not?

2) I'd try solving the confirmed class C once we fix the unconfirmed case.

matiasz8, 2018-01-04:
At the moment, I has not receive any frame in RX2. I'll decouple this issue.

gotthardp, 2018-02-27:
Hello and sorry for the delay. For me the Class C works. Were you able to succeed here, please?

matiasz8, 2018-02-27:
Thanks for answer, no problem. I hadn't could transmit any packet by Class C at the moment, but I'll try to test again.

gotthardp, 2018-02-27:
Thanks. Please try the latest 0.5.x, if possible.

matiasz8, 2018-02-27:
Ohh, ok, thanks for that.

matiasz8, 2018-03-14:
Hello, I'm trying to test this again, but in this case, I haven't a WebSocket application to select from the combo box. In this case, need I create an Earlang file or just with create a new connector to websocket server It is done?

gotthardp, 2018-03-14:
Create a backend handler (this is your application name) and then link it with a device profile *and* your WebSocket Connector.

matiasz8, 2018-03-15:
Hello again,
I can't create a connector webSocket. I has used the examples:

are them all right?

ws://localhost:8080/ws/events/00020507
ws://127.1.1.1:8080/ws/events/00020507
ws://10.0.0.101:8080/ws/events/00020507

[image]

[image]

[image]

gotthardp, 2018-03-15:
Follow the documentation :-) URI shall be 'ws:' only. The rest of the url goes into the topic.

gotthardp, 2018-03-19:
Make sure there are no Failed flags. When a Failed flag is raised, nothing works.

matiasz8, 2018-03-19:
MMm, I changed It:
I put into the field "Failed" right now:
 - Nothing (1º time)
 - Badarg (2º time)
 - Network (3º time)
 - topic (4º time)

But, the button show the same behavior.

gotthardp, 2018-03-19:
The Failed field shall be empty. If the server raises a Failed flag the debug.log contains more details.

matiasz8, 2018-03-19:
Look at this:
If I empty the values of "Failed", when I return to the list, the fields have two "badarg" in the field.
And then I check the connector again and actually have two "badarg" again.

[image]
[image]
[image]

gotthardp, 2018-03-19:
For the WS: connector the Publish Uplinks/Events shall start with the backslash (as the documentation says).

matiasz8, 2018-03-19:
Is it correct this way?
[image]

gotthardp, 2018-03-19:
Looks better. Remove the Failed flag and try if the server accepts it.

gotthardp, 2018-03-31:
Any progress?  May I  close the issue?

matiasz8, 2018-04-04:
Hello again,
I was able to complete the sending of the data from my LoraWan module to the WebSocket demo.html. But I could not reverse it.

I sent this json through the ws.html, but the Dashboard informs me that the Downlink was lost.

[image]

[image]

Another thing that happens to me is that when I ask if the module has packages to receive, it returns a data, but when converting it, it does not have a conversion to Ascii.

Do I need to pass it through a converter with a seed for it?

[image]

matiasz8, 2018-04-20:
Hello again, 
I've found a bug, or maybe It's a bad configuration that I've set.

I've been debuging the Package Forwarder, and when I send a packet througth the ws.html the Package forwarder log show me that the packet was reject because a bad frequency.

Look, this is the message that I sent:
[image]

And this is the error: the frequency is equal "9", so the pkt fwd say that this value must be between 902000000 to 928000000:
[image]

gotthardp, 2018-04-20:
Feels like you misconfigured "RX2 Freq" in the Network or Profile settings.

matiasz8, 2018-04-25:
Hello,

Thanks for that. I fixed that problem, I set wrong my RXWin2 Spread Factor, and the global_conf.json configuration was wrong too.

For the tests, I was need to add the parameter "confirmed":true. Without this parameter the module not recieve the message. So, this is the message that I was sending:

Class C:
{"data":"30", "port":12, "time":"immediately", "confirmed":true}

I'm now at the beginning of the test: I set the TX Windows Auto. 

I send the message through the ws.html. This only arrive if I send another message: just like the behavior of Class A.

The message arrive thougth RXWin1 at the moment. I know that the class c message arrive

 on RXwin2 or RXwin0, but at the moment I can't get that result.

I attach my configurations:

[image]

[image]

[image]

the result:

[image]

[image]

gotthardp, 2018-04-25:
Why is "Used RX2 DR" = SF8 500? This is a non-standard value, which may cause the troubles.

matiasz8, 2018-04-26:
Mmm, I was checking what packet am I receiving as downlink from packet forwarder.

It's true that when the downlink arrive to the packet forwarder the "datr" received is "SF8BW500"

That data: Used RX2 Freq, where can I set?

My module's setting is 923300000, DR8, that corresponding to SF12/500kHz.

I set the RX2 in the application with SF12/500kHz, as you can see in the previous captures.

This is the capture of my module:

[image]

gotthardp, 2018-04-26:
What does your network ADR settings say? You should align the RXWIN2 settings in your device with the "Used RX2 DR". It will not work otherwise.

matiasz8, 2018-04-26:
Great! Yes, my mistake was setting in Network - ADR, the RXWin2 with SF8 500.
Thank for all!

closed 2018-04-26 by matiasz8

gotthardp, 2018-04-27:
Cool! We finally made it 😃