gotthardp/lorawan-server #268: TX Windows RX2 not receive package confirmation

matiasz8, 2018-01-04:
Hello, as I said before,

I has configure the node as you can see at the picture. 

I have performed the following test:

I has change TX Window to "RX2" and I's tried to receive the acknowledge from a package sent but nothing happends. 

Thanks

I have the release 0.4.13
I have this in the debug file: 
2018-01-04 15:39:11.914 [warning] <0.291.0> gateway 0000112233445500 {ack_lost,82}
2018-01-04 15:39:20.079 [error] <0.297.0> node 00020505 {fcnt_gap_too_large,4}

[image]

gotthardp, 2018-02-27:
Hello. I set a node to RX2, sent a confirmed downlink (class A) and received an ACK. To me this works.

matiasz8, 2018-02-27:
Thanks for answer. Ok, I'll try to find if I have some wrong at the module's startup. I tested with Class A and Class C too.

matiasz8, 2018-03-14:
Hello, thanks for that. I'm testing with 5.3 versión and It's Works.

closed 2018-03-14 by matiasz8