gotthardp/lorawan-server #269: Question

hallard, 2018-01-09:
Hi Petr,
Just may be a stupid question, does a device/node can works with both websocket and backend?
For example have one app connected to server with websocket and the other one with MQTT and have on each device received packet, data pushed on Websocket AND MQTT ?
If so how can  achieve this?
Thanks

gotthardp, 2018-01-09:
Hello. Not possible for now, but it is possible in the 0.5.0-- there you can assign multiple "Connectors" to the same "Handler" and therefore forward the application frames via various protocols.

hallard, 2018-01-09:
Thanks for this quick answer,
To test 0.5 version I need branch `devel`, correct? Install as usual?

gotthardp, 2018-01-09:
Yes, `devel`. It should work, but be aware there may be more bugs that usually and that you need to review the entire configuration as many fields have been moved.

hallard, 2018-01-09:
Thanks, I'll take care of that.

closed 2018-01-09 by hallard