gotthardp/lorawan-server #274: websocket issue on ios devices

hallard, 2018-01-10:
Petr,
During all my POC concept I've done a websocket HTML page (based on your `ws.html` demo) which is working fine on my desktop. My GW is a RPI 3 with ic880a concentrator. The GW is configured as  WiFi AP and webserver (nginx) for html pages.

So that said, my computer connected to WiFi of GW and chrome browsing `myws.html` 
Each time lora packet is received I got it on websocket. everything is amazing (thanks for your lorawan server)
Now I'm connecting with iPad Mini or iPhone 5 SE (both ios11) and I reveice very often websocket packet delayed up by 10 to 30 seconds and sometimes even get disconnected. I suspected Safari so installed chrome on my iPhone, exactly same problem. In the meanwhile chrome on my desktop is receiving all packets immediately.
Are you aware of this kind of issue? Could it be a problem with lorawan-websocket and iOs?
I need to check other websocket (server side) implementation to check where it comes from.
Anyway, good to know this problem. I'd like to test with browser MQTT JS but it's also based on websocket.
If you have a magic idea, I'll take it

gotthardp, 2018-01-10:
I have no magic idea, but I'd suggest to run Wireshark, inspect the HTTP/WS traffic and see at which moment the delay is created. This could then help identifying the issue.

hallard, 2018-01-11:
Thanks,
I'll check this on RPI with tcpdump.

closed 2018-01-11 by hallard