gotthardp/lorawan-server #276: Newbie help: single channel 433MHz packet forwarding help

kelll27, 2018-01-14:
Hey, so I am newbie when it comes to LoRa and LoRaWAN and i would like some help
Currently I am trying to set up a DIY node and Gateway, using Adafruit 96W (433MHz, single channel) modules. 
I am today becoming familiar with the use of private servers for LoRa and have seen i need to confiure the pack forwarding with 433MHz band. Can anyone help me configure the necessary files and point me to resources to learn to code for the 433MHz band?

gotthardp, 2018-01-14:
Hello. I am not able to help with a setup of a single-channel packet forwarder.

kelll27, 2018-01-14:
Oh ok, thank you anyway

gotthardp, 2018-02-03:
Closing due to inactivity. The questions are better to ask on the mailing list.

closed 2018-02-03 by gotthardp