gotthardp/lorawan-server #278: ADR Graphs have upside down Axis for some values

waynestallwood-kcom, 2018-01-15:
Hi, Running latest Docker Image. 

On the RX graph in The plotted lines on the page, admin#/nodes/edit/ABC12345 seem to have an inverted axis for Power dBm, 14dBm appears at the bottom of the axis and 2 appears at roughly the midway point on the graph.

Should this not be inverted to match the displayed axis on the lefthand side of the chart ?

Repeated on Safari Browser and latest build of Chrome for OSX

gotthardp, 2018-01-15:
Hi. The graph displays an "index" of DR or Power as defined in LoRa standard and DR=0 is lowest DR, but Power=0 is highest power. I admit it is confusing though. Will try to invert it.

waynestallwood-kcom, 2018-01-15:
I wonder if the true dBm is the best axis, the others are confusing and conflicting depending on where you are it seems. Your graph would work on US frequencies it seems ?

 *        EU     	433 MHz		  US
 *    0:  N/A        10 dBm       N/A
 *    1:  14 dBm      7 dBm       N/A
 *    2:  11 dBm      4 dBm       N/A
 *    3:   8 dBm      1 dBm       N/A
 *    4:   5 dBm     -2 dBm       N/A
 *    5:   2 dBm     -5 dBm       20 dBm
 *    6:  N/A        N/A          18 dBm
 *    7:  N/A        N/A          16 dBm
 *    8:  N/A        N/A          14 dBm
 *    9:  N/A        N/A          12 dBm
 *    10: N/A        N/A          10 dBm

gotthardp, 2018-01-15:
There are 3 lines in that graph: frequency, data rate and power, but the graph allows only two axis, so two lines of them have to use a common axis. That's a technical limitation. I will try a trick to put DR and Power on the same axis, but use the true dBm as a unit.

gotthardp, 2018-01-15:
the master docker image shall have the power inverted as you requested

closed 2018-01-15 by gotthardp