gotthardp/lorawan-server #280: Multitech mLinux gateway

jubbs, 2018-01-16:
Has anyone setup a Multitech mLinux gateway to connect to this server? Do you have any troubleshooting tips. What gateways has it been used with?

gotthardp, 2018-01-16:
Yes,  Multitech mLinux gateway was connected to this server. You need to convert it to basic packet_forwarder as described here: http://www.multitech.net/developer/software/lora/conduit-mlinux-convert-to-basic-packet-forwarder/

jubbs, 2018-01-16:
I have the packet forwarder running on my Gateway. 
The lorawan-server on the same network.
I have only changed the IP of the server to mine in the global_conf.json
In the lorawan-server I have the MAC set to the EUI of the Gateway, NetID 000000, SubID <<empty>>, TX Chain 0, TX Power 14, Gain 0

Any suggestions?

gotthardp, 2018-01-16:
That sounds good. Doesnt it work?

jubbs, 2018-01-16:
Not getting anything.

My gateway packet forwarder is running in top but I am not getting any log files so I will keep investigating.

gotthardp, 2018-01-16:
Besides global_conf.json sometimes there is also local_conf.json that overrides the global. And sometimes there are various instances of the config, but only one of them is used.
The server should report it is seeing the gateway, so if server does not receive anything, then either the IP is misconfigured, or there is a firewall.

Also, check the server port number in the gateway conf. Some other networks are using different port numbers than the default packet_forwarder.

jubbs, 2018-01-16:
What port should I be using?

gotthardp, 2018-01-16:
By default my server (and default port-forwarders) use 1680.

jubbs, 2018-01-16:
Sorted 
The packet forwarder on the Multitech gateway defaults to 20000. Changed it to 1680.
Connecting now. Thanks.

gotthardp, 2018-01-16:
Cool! I am glad it works now.

closed 2018-01-16 by gotthardp

jubbs, 2018-01-16:
Ports were the issue 1680 as opposed to 20000

Thanks