gotthardp/lorawan-server #283: List handling

gnob417, 2018-01-19:
Hi, I'm so sorry about questioning like language stuff.
But I'm struggling with the erlang because of "bound, unbound" thing.

here's the thing.

i have a list of stored_frames (S_list).

1. i want to make a list (let me call Dev_list) with DevAddr of stored_frames from S_list. 

2. If Dev_list from 1th step is [7, 2, 2, 3, 4, 3, 5], i want to extract the list like [7, 2, 3, 4, 5]
    which is the list having each address once. (sequence is not that important)

How do I have to overcome this step? 
unchangable lists are hard to handle. 
give me an advice.

thank you so much.

gotthardp, 2018-01-19:
http://erlang.org/doc/man/lists.html#usort-1 may be the answer?

gotthardp, 2018-01-19:
And to do the "extraction" there is http://erlang.org/doc/man/lists.html#map-2

gnob417, 2018-01-19:
@gotthardp 
You're wonderful. I figured it out
Thank you so much again :)

closed 2018-01-19 by gotthardp

gnob417, 2018-01-21:
@gotthardp 
Hi, i have one more question!

If i have a list of 4 elements and i want to make them to bit data structure.
let list name is "List =[A, B, C]", and there is one variable D.
How do i make it to "<< D:32, A, B, C >>" ?

gotthardp, 2018-01-21:
How about
```erlang
BinList =
  lists:foldr(
    fun(Elem, Acc) ->
      <<Elem, Acc/binary>>
    end,
    <<>>, List),
<<D:32, BinList/binary>>
```

gotthardp, 2018-01-21:
If there are always 3 elements, then simple
```
[A,B,C] = List
```
works too

gnob417, 2018-01-22:
In your BinList example code,
What if elements of the "List" A, B, C is each a bit element? 
(like List = [<<1,22,33>>, <<2,3,4>>, <<4,5,1>>])

gnob417, 2018-01-22:
Thanks a lot!
I'm curious about how to limit the size of "BinList".
I mean, If the number of  elements in "List" is over 8, I want to limit the number of BinList to 6.
is that possible?

gotthardp, 2018-01-22:
Then cut it: http://erlang.org/doc/man/lists.html#sublist-2