gotthardp/lorawan-server #284: shall_reply stays true

vvanbeveren, 2018-01-22:
I have a single mote and a single gateway connected to the LoRaWAN server/ I created a simple python script to log all data from any of the motes. Now I notice a flag in the JSON from the server: `"shall_reply" : true`. This flag is present in the first message. Then it is gone for a number of messages (about 18), and then it reappears and never goes away. 
I noticed I can send data to the mote , which is just receives, so the RX window seems to work.

Should I be worried about this? What does this flag indicate?

lora-wan server version 0.4.3 
Node is configured with ADR.

gotthardp, 2018-01-22:
This is not necessarily a bad behaviour. The "shall_reply" indicated that server will have to respond to the uplink no matter whether a downlink is scheduled or not.

Could you send me the server debug.log to check if the behaviour is correct?

vvanbeveren, 2018-01-26:
I've switched to using ADR. For a stress test the node is sending on SF7 quite often, but not as often to exceed EU regulations. At some point the server will send a 2nd ADR message. From that point on the RN2483 gives no_free_ch for a very long time.

[debug.log](https://github.com/gotthardp/lorawan-server/files/1667579/debug.log)

vvanbeveren, 2018-01-26:
Oh, and I can send messages to the node. So, the RX path seems to work.

gotthardp, 2018-01-26:
I know what happened. The device transmitted at SF7/125 and then Auto-ADR switched to SF7/250, which is not supported/allowed on the default channels, so the device didn't obey the request and server kept requesting it (hence shall_reply stayed true).
This behaviour was "somewhat" mitigated in 0.5 (where you can limit max DR), but it remains in 0.4.x.

gotthardp, 2018-02-03:
Just fixed.

closed 2018-02-03 by gotthardp