gotthardp/lorawan-server #285: Bad behaviour when two gateways in range during join ?

waynestallwood-kcom, 2018-01-22:
Not sure where this bug lies but I think I'm seeing a problem when two gateways are in range during a OTAA join. 

When this happens I see two Join's from the same DevEUI a few seconds apart

  | info | 2018-01-22 12:18:36 | 2018-01-22 12:18:36 | 1 | device | 70B3D549904FAC42 | join | <<"ABC54321">>
  | info | 2018-01-22 12:17:44 | 2018-01-22 12:17:44 | 1 | device | 70B3D549904FAC42 | join | <<"ABC54321">>

Then every subsequent message has bad_mic. Perhaps because the NwkSKey has been rotated for each join and now my device has the wrong one ? It's not 100% repeatable (perhaps because sometimes I end up with the 2nd key ?) Unfortunately I'm unaware of a way to get the current NwkSkey from my device to verify.

Is that possible...the behaviour during uplinks if two gateways are in range seems reasonable in that I see two messages with the same fcnt but only one goes through the backend. But should I expect any problems when two gateways forward a join packet ?

If this is a real problem rather than an issue with my devices or setup then I wondered if a potential workaround would be to ignore a second join request (if received from the same DevEUI) say 5 seconds after a successful one ?

gotthardp, 2018-01-22:
Which server version?

waynestallwood-kcom, 2018-01-22:
latest container as launched 2 weeks ago. Actually I'm starting to think it's device side as I just noticed the log entries are further apart than I thought on the timeline, so unless this is verified by someone with a different device (I'm on a LoPy) then I'd park it sorry :)

gotthardp, 2018-01-22:
One way is to check the gateway log. If the frame was received twice on each gateway, then it's a device issue. If the packet was received only once on each gateway, then it's a server issue.

closed 2018-02-02 by gotthardp