gotthardp/lorawan-server #288: v0.5.0 websocket and app

hallard, 2018-01-25:
Hi Petr,
I just tried my web app (with websockets) with v0.5.0 but there are some change that I can't figure out
- before device would belong to a group and we could connect with websocket to that group of devices. Today I imagine it's related to profile?
- If so, on profile config on application I can just select semtech-mode not websocket ?

Did I missed something ? How can I connect with websocket on a group of devices as before ?

Thanks

hallard, 2018-01-25:
You're welcome 
Thanks got it, 
Do I need a connector for websocket or handler will do the job?
Anyway there are a lot of new parameters to set and on some may be put in documentation some default "out of the box" (same as default for V4) that works, because on some I have no idea what to put in to get it working
Here what I put for a Network of EU 868, not sure all is good, but your validation could be a good start
**on general**
[image]

**On network **
Initial Channels I put => 0-2
**On ADR**
Not sure what put put
[image]
Initial RX2 Freq (MHz) => no idea what to put in this one, the other too.

Another strange thing, I created a handler
[image]

but it's not showing on application (even semtech-mode has gone ?!?)
[image]

Anyway I need to get things working today, so don't worry about this for now, I'm going back to v0.4.13 until I have more time to test in real the v0.5.0,
May be I can try v0.4.13, restore with dbimport and then launch the upgrade to v0.5.0 just to see ;-)

gotthardp, 2018-01-26:
I thought the Network settings could be a pain. I added a new file with some suggestions, so it can work more "out of a box".
https://github.com/gotthardp/lorawan-server/blob/master/doc/Regions.md

gotthardp, 2018-01-26:
Also, I cannot reproduce your trouble with application not showing in the list. Could you please have a look on the Javascript console in your browser to see if there is some error? (In Firefox it's under Web Developper -- Web Console.) Or perhaps debug.log shows something.

aymeri06, 2018-01-31:
hi everyone, i have the same problem that hallard, i don't have websocket in choice for application with the new version (5.0) it's bug ?

thanks

gotthardp, 2018-01-31:
Hello. Please read my answer above. There is no 'websocket' application in 0.5.0. You need to create a websocket handler+connector for "ws:" as described in the documentaton.

aymeri06, 2018-02-01:
sorry i don't see your response ^^ but thank you it's good

gotthardp, 2018-02-02:
Clarified.

closed 2018-02-02 by gotthardp

nupik, 2018-02-07:
Still can't create WebSocket connection 
[image]
[image]
[image]
[image]

gotthardp, 2018-02-07:
And what error are you getting?

nupik, 2018-02-08:
Just have no connection by URI( /ws/uplink/010A8AA2 )
[image]

gotthardp, 2018-02-08:
Oops. I confirm this is a bug. I just fixed that in the 'main' branch.

hallard, 2018-02-08:
@gotthardp 
By the way, about websockets, I noticed in latest v0.4.x  that when a websocket client send data (in my case json) and the data is not correct (example unexisting node or bad format) the server disconnect the client. I do not know if it is the expected behavior or not, but this mean client can't do anything until reconnexion (that can be async).
So my question is:
Is it the same way on v0.5.x? and if so would it be possible that the server just discard the data instead of closing websocket?
Thanks

gotthardp, 2018-02-08:
@hallard yes, this behaviour is also in 0.5.x, but I can consider changing it.

nupik, 2018-02-09:
Yes, I think better just send error report back. It will be more robust

nupik, 2018-02-09:
Next step. Question: connection is ok, but I doesn't receive any data by websocket.
[image]
[image]