gotthardp/lorawan-server #289: Question: uplink parsing + intercepting + discard message

rayhwang-kcom, 2018-01-25:
Hi,

I am wondering if there's a way to intercept uplinks and stop it being sent on? 

We are sending a single space in order to trigger a downlink reception however we do not want this to be sent to AWS. Is there a way to parse and intercept and discard it at the server level using uplink parsing?

Thanks in advance.

gotthardp, 2018-01-25:
Hello. Yes, there is (undocumented, I admit). The documentation says the Parse Uplink function returns a field map (e.g. `#{field => "value"}`. See https://github.com/gotthardp/lorawan-server/blob/master/doc/Handlers.md#parse-uplink

The Parse Uplink is however also allowed to send a list of the field maps to send multiple backend messages based on one frame -- if you send an empty list on the other hand, no uplink message will be sent.

Long story short: write a Parse Uplink function which in some cases returns an empty list (`[]`). Then the frame will be stopped.

gotthardp, 2018-01-25:
Of course, this is valid for the version 0.5.0 only. In previous 0.4.x this is not possible I am afraid.

rayhwang-kcom, 2018-01-25:
Ok - I will see if it's possible to roll out 0.5.0.

gotthardp, 2018-02-02:
Clarified in the documentation.

closed 2018-02-02 by gotthardp