gotthardp/lorawan-server #291: how to do downlink message

Mels67, 2018-01-25:
Hi,

I've configured this LoRaWAN server with a handler and MQTT connector. Receive topic is IOTGW/IN/{devaddr}
If I use MQTT tool I can publish a message to IOTGW/IN/03F5FFC8/  and I publish {"data":"01"}
On this same MQTT tool which is also subscribed to IOTGW/# I see this message being published. On the nodes menu LORAWan server I don't see any downlink message in que, nothing mentioned after downlinks:  What do I do wrong? of course the device is connected and working (sending a message when an input is triggered). But I assume it goes wrong before, since I don't see any downlink que on the LoRaWAN server in the menu nodes (and selecting the right node of course). Thanks!

best regards, Mels

gotthardp, 2018-01-25:
Hello. What "received topic" you did set? There should be some target device identifier, either in the received topic or in the message itself.

gotthardp, 2018-01-25:
Hello. What server version 0.4 or 0.5?

gotthardp, 2018-01-25:
Also, is there anything suspicious in server's debug.log?

Mels67, 2018-01-26:
Hi Petr, I did do it ok but needed to restart the server (which I didn't) now I see the downlink message also on the node menu on the server and the device also responds! Very good! I'm still on the latest 0.4 for the moment. Will try and test 0.5 later. Thanks for your fast response!

closed 2018-01-26 by Mels67