gotthardp/lorawan-server #292: How set an APIKEY in the description of an http/https connector ?

ogattaz, 2018-01-25:
We have to set the Authorization header with a revocable JWT token 
eg.
POST /services/rawdata/measures HTTP/1.1
...
Authorization: Bearer eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJzdWIiOiI1YTI4MTkwYTdhNjhiYTdmYmQ2MjNjMjQiLCJraW5kIjoiQUNDT1VOVCIsImlhdCI6MTUxMjU3NzMzNywidGlkIjoiNWEyODEwZjQ3YTY4YmE3ZmJkNjIzYzFkIiwianRpIjoiNWEyODE5Mzk3YTY4YmE3ZmJkNjIzYzI1In0.OO79ZS0BDO12PaQjlp2_5jO7tOUirGfl-ynZ0zWImV8
...

Where can we store this information to be used as the content of the Authorization header ?

Regards

gotthardp, 2018-01-25:
This kind authorization is not supported at this moment, I am afraid.

gotthardp, 2018-08-14:
I just implemented this for 0.6.0. See the authentication type "Header+Token".

closed 2018-08-14 by gotthardp