gotthardp/lorawan-server #294: Time difference

gnob417, 2018-01-29:
Hi,
I want to calculate the time from a certain moment to time when a packet is transmitted.
a certain moment is changed periodically, and the source code in which i can get packet transmission time is different erlang file from the file which i can get certain moment certain moment time. 

I know there is `timer:now_diff(T2, T1)`, however, the each T2 and T1 from above explanation is located in different function. (there is no global variable in erlang as far as I know..) 

How can I implement this kind of situation?

gotthardp, 2018-01-29:
Hello. If say T1 is the packet transmission. What is your T2?

gnob417, 2018-01-30:
T2 could be another packet transmission time.

gotthardp, 2018-02-03:
You need to send both to the same process or store both in a single database table.

closed 2018-02-17 by gotthardp