gotthardp/lorawan-server #296: JSON message via MQTT

webfrank, 2018-01-30:
Hi,
   with latest docker version 0.5.0 I cannot receive all the selected fields with JSON.
I configured the handler to send devaddr, fcnt, port, data, datetime and all_gw but I receive only all_gw and decoded data.
[image]
[image]

webfrank, 2018-01-30:
Ok, I found it was changed the parse uplink function, changing it worked.

closed 2018-01-30 by webfrank

gotthardp, 2018-01-30:
Yeah :) You should not even receive the all_gw :) The syntax changed a bit-- the _Port parameter for Uplink changed to Fields and disappeared for Downlink. Also, when adding custom attributes you should amend the Fields by `Fields#{temp => Temp, irms => Irms}`.