gotthardp/lorawan-server #297: Unexpected_data Applictaion error - Please help.

netuserlv, 2018-01-30:
Hey! I`m running the the server and I have connected my gateway to the server and everything is running just fine until I`m sending the data to the application. I know that the application is not configured right for my node. But it is all I have maybe someone could help me configure the application so that It would work for my node. I`m using sodaq mbili board with the rn2483 chip on it. And simply using the ATT_lora_iot examples found on github. I`m really not that knowing that in LORA so thats all that works for me. Maybe someone could help me building the erlang applications for sodaq mbili node and the att_lora_iot examples. I`m using Nemeus pico gateway.

gotthardp, 2018-01-30:
Hello. And what do you want to do with the data from the device? Display them, store them, process them?

netuserlv, 2018-01-30:
For now I guess just store them. I think I could figure out the displaying part myself.

gotthardp, 2018-02-03:
I recommend you to use WebSockets, i.e. connect to the server to receive all uplinks and handle them as you like. You should review the Handlers and Connectors Guide (see /doc) to understand how to do that.

netuserlv, 2018-02-06:
Ok maybe could You please help me with that.... I configure my connector to 
URL = "ws:"
Publish Uplinks = "/ws/uplink/00441018"
Publish Events = "/ws/events/00441018"

and when i go to the link = "http://localhost:8080/admin/ws.html"
I cant do anything or am I doing something wrong because, what is the path to connect to?

gotthardp, 2018-02-06:
the path is what you entered: `/ws/uplink/00441018`

netuserlv, 2018-02-06:
Yeah i tried that, but the connection doesnt do anything :/

netuserlv, 2018-02-06:
and it shows me in the server http  error "/ws/uplink/00441018" doesnt exist

gotthardp, 2018-02-06:
you are supposed to enter the URI to the ws.html

netuserlv, 2018-02-06:
so I have to replace in the connector url "ws:" to "http://localhost:8080/admin/ws.html"

gotthardp, 2018-02-06:
No. Create a connector, set URL to "ws:", enter the PublishUplinks as you did. Then, use a web browser to open http://localhost:8080/admin/ws.html and THERE enter the Path you used for PublishUplinks.

netuserlv, 2018-02-06:
Yep, that doesnt work :/ I do all that when I make the connector with those settings as I did, then I go to the ws html page insert the path and press connection nothing happens :/ Btw Maybe I`am sounding a bit ungrateful, but that is not the case, You are the best for helping me and making this server :)

netuserlv, 2018-02-06:
I`m using windows maybe that is the problem ?

gotthardp, 2018-02-06:
Send me a screenshot of the Connector and of the ws.html after you clicked "Connect", please

netuserlv, 2018-02-06:
[image]
[image]
[image]

gotthardp, 2018-02-06:
Did you click "Connection" on ws.html? There is supposed to be some text.

netuserlv, 2018-02-06:
yep i pressed the connect :/on the ws.html

netuserlv, 2018-02-06:
tried to change the types to nothing works....

gotthardp, 2018-02-06:
that appears to be the problem. Would you mind opening a javascript debug console for ws.html?

netuserlv, 2018-02-06:
[image]

I hope this is the one

gotthardp, 2018-02-06:
Aaaaaah. The connector has a "Failed" flag set to "topic". You should clear that flag.

netuserlv, 2018-02-06:
Ammhhh ok. Do that is in the server or in the html file, I kinda don`t know how to do that :D

gotthardp, 2018-02-06:
in the web-admin for Connector. There is a little cross after the "topic" flag. Click on that and then save the Connector.

netuserlv, 2018-02-06:
ouhhhhh :D

netuserlv, 2018-02-06:
Now i get text :D

netuserlv, 2018-02-06:
I hope I wont have to bother you anymore :) big thanks from Latvia

gotthardp, 2018-02-06:
As a next step you should change the Publish Uplinks = "/ws/uplink/{devaddr}" (without quotes) so that it works for any DevAddr

netuserlv, 2018-02-06:
Ill do that :) any other tips?

gotthardp, 2018-02-06:
No. Try and see :)

closed 2018-02-17 by gotthardp