gotthardp/lorawan-server #299: Decode payload of test device (Parse Uplink)

kazi93, 2018-01-31:
Hi! Greetings. @gotthardp @altishchenko 
I would like to decode a test device. However, nothing show up in the filed :(
Data : for example
9F19523224300132376037170610081606
[image]
[image]

Parse Uplink : 
fun(Fields, <<Status, temperature:16, Lat:24, Lon:24, vdd, rssi, snr>>) ->
  Fields#{status => Status, temp => temperature/100, batt => vdd, rssi => rssi, snr => snr}
end.

or 

Parse Uplink : 
fun(Fields, <<16#01, Status, temperature:16, Lat:24, Lon:24, vdd, rssi, snr>>) ->
  Fields#{status => Status, temp => temperature/100, batt => vdd, rssi => rssi, snr => snr}
end.

gotthardp, 2018-01-31:
The variables have to start with an Upper Case letter. Also, use the correct field length.

```erlang
fun(Fields, <<Status, Temp, Lat:32, Lon:32, Qualite, UL, DL, Vdd:16, Rssi, Snr>>) ->
  Fields#{status => Status, temp => Temp, batt => Vdd, rssi => Rssi, snr => Snr}
end.
```

altishchenko, 2018-02-06:
@kazi93 I am sorry, I missed this one! Are by any chance using something like Adenius tester? Be warned then, that not all the fields are present in the data. There presence is determined by the first byte. If it is some sort of Adenius, I'll send you a function I use to parse it.

gradzik, 2018-02-16:
Hello.

I have a similar problem with parse uplink.

Recieved Frame: 58676A68
Connector sends it as a row data and application (Azure) recieve 4 bytes: Xgjh ('X' -> 0x58 etc.)

How can I parse it in handler to receive 8 bytes string ("5866F848")?

gotthardp, 2018-02-16:
@gradzik, if you switch to JSON the data get converted, Can you switch to JSON?

gradzik, 2018-02-16:
I have to be compatible with another devices (not lora) and beacuse of that i can't use JSON... Is that conversion very problematic?

gotthardp, 2018-02-16:
Not really, but it's more verbose. To convert the entire payload to a hexastring you can do
```erlang
fun(Fields, Payload) ->
  Fields#{data => lorawan_utils:binary_to_hex(Payload)}
end.
```
(I assume the connector format is **Raw*)

gradzik, 2018-02-19:
It's what I needed, thank you :)

closed 2018-02-19 by gotthardp