gotthardp/lorawan-server #301: build_failed error when parsing a downlink message

Neuromystix, 2018-02-05:
I'm trying to parse a downlink message using the example:
```
fun(#{led := LED}) ->
  <<LED>>
end.
```

In my connectors, with a mosquitto broker, i have configured the following topics:
```
Subscribe          downlink_nodes/#
Received Topic     downlink_nodes/{deveui}
```

I can send downlink messages, publishing to `downlink_nodes/`  a message like: `{"deveui":"AABBCCDDEE123456","data":"0123", "confirmed":true}`
However, when trying to publish to `downlink_nodes/AABBCCDDEE123456` a message like 
 `{"fields":{"led":1}}` 
i'm allways getting a _build_failed | {error,function_clause}._ 
Am i doing it wrong or it's a bug?

Best regards

gotthardp, 2018-02-05:
Hello. The function you wrote expects a field named "led", which you didn't send, hence the error. You don't need any build function if you are just sending raw data.

Neuromystix, 2018-02-05:
Right, my bad. Just published `{"led":1}` and it worked fine. Thanks.

closed 2018-02-05 by Neuromystix