gotthardp/lorawan-server #304: lorawan-server app does not start ( Mututech-mlinux3.3.13)

pdl86, 2018-02-12:
Hello

we are trying the 0.5.0 version on a Multitch mLinux gateway.
the compilation seems to have gone through. ( following Multitech and your documentation)
but after having flashed the gateway the lorawan server does not start. 
we got an error , as erlang does not find the lorawan_app "source" : 

trace taken , while launching the lorawan server in foreground : 

/usr/sbin/start-stop-daemon --start -u lorawan --make-pidfile --pidfile /var/run/lora/lorawan-server-1.pid --startas /bin/bash -- -c "LORAWAN_HOME=/var/lib/lorawan-server LORAWAN_LOG_ROOT=/var/log/lorawan-server /usr/lib/lorawan-server/bin/lorawan-server"

[os_mon] memory supervisor port (memsup): Erlang has closed
[os_mon] cpu supervisor port (cpu_sup): Erlang has closed

{"init terminating in do_boot",{{badmatch,{error,{xmerl,{"no such file or directory","xmerl.app"}}}},[{lorawan_app,start,0,[{file,"/home/transnumeric/mlinux-3.3.13/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.5.0-r0/git/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,13}]},{init,start_em,1,[]},{init,do_boot,3,[]}]}}

init terminating in do_boot ({{badmatch,{error,{xmerl,{_}}}},[{lorawan_app,start,0,[{_},{_}]},{init,start_em,1,[]},{init,do_boot,3,[]}]})

any ideas? 
thanks in advance 
best regards
Philippe

pdl86, 2018-02-12:
Hello 

we have "downgrade " to 0.4.12 and it works : aka with the same compilation and image generation processes, the gateway start the lorawans-server 

So it seems the issue is linked to 0.5.0 

regards
Philippe

pdl86, 2018-02-13:
Hello 
it looks like the xmerl dependency is missing somewhere  ( for mlinux) 

thanks in advance 
regards
Philippe

gotthardp, 2018-02-13:
And could you install the `erlang-xmerl` or use the updated `lorawan-server_0.5.0.bb` and build the package again, please?

pdl86, 2018-02-13:
Hello 
Ok , we have tried i , and got a "strange" error : 
while executing 

bitbake mlinux-lorawan-image : 
ERROR: Nothing PROVIDES 'mlinux-lorawan-image'. Close matches:
  mlinux-mono-image
  mlinux-base-image
  mlinux-factory-image

Seeing that , we have completely clean the environnam ( totally erase teh mlinux-3.x folders
and restart from scract ( gut the mlinux distrib, then erlang, then lorawan , modified conf/bblayers.conf,   the the ..... meta-lorawa-server/recipes-mlinux/images/mlinux-lorawan-image.bb   
in order to include gpsd and gps-utiles ( save time later one) 

and got exactly the same issue .

not being very familiar with bitbake, we are stuck .

any help appreciated.
best regards
Philippe

gotthardp, 2018-02-13:
I believe you skipped the third bullet from https://github.com/gotthardp/meta-lorawan#usage-instructions

pdl86, 2018-02-13:
Hi

you are completely right !  ( and one of my team member has mentionend
it !)

but, this is not mentioned in this instruction :

https://github.com/gotthardp/lorawan-server/blob/master/doc/Embedded.md  
, which is the one we have followed.

so better to have it corrected.

Anyway it works, and image generation is ont ist way ( few hours) .

I will keep you updated tomorrow

best regards

Philippe

gotthardp, 2018-02-14:
Yes, this is my fault. I just fixed that. Thanks for reporting it.

closed 2018-02-14 by gotthardp

pdl86, 2018-02-15:
I confirm it works now  ( with the xmerl "correction", in bb file) 
thanks

gotthardp, 2018-02-15:
Cool :) thanks for testing it.