gotthardp/lorawan-server #306: Change the Dowlink payload size

javobarra, 2018-02-16:
Hi @gotthardp !
Is it possible to change this size?
I used your Websocket Client for send messages with different sizes. When i send messages with more than 50 bytes, my end device never receive the dowlink message.
I think the limitation is in your server and not in the End Device. Do you know if is it possible to change that?
Thanks in advance!

Javo

gotthardp, 2018-02-16:
Hello. The limitation is in the LoRaWAN protocol itself. Use a higher Data Rate and you will be able to send more data. Have a look in the LoRaWAN specification, at the "Maximum Payload Size" table for your region.

javobarra, 2018-02-16:
Dear @gotthardp ,
In spite of Im using a data rate 3  (SF7  US region) for the uplinks which has a MAX payload of 250 bytes, i couldn't pass the 55 bytes for the downlink messages. Lorawan answer it with a data rate which depends on the uplink message, in this case dr10 (Max payload: 230bytes), but this is not my case.
Apart of that, those limitations are for a public loranwan network. What happened if i have a private network? I could modify the uplink payload size, because i use lmic library so i change some register values. But now i want to change the size for the downlink packets, so is it possible to change it in your lorawan server?
Thanks for your help!

gotthardp, 2018-02-16:
The server does not impose any size limits on the payload. Please check the server and gateway logs for any errors or warnings that could have caused this.

javobarra, 2018-02-20:
@gotthardp , thanks for your help!
I had to modify LMIC library and i could transfer payloads bigger than 55 bytes.
For everyone,  i changed "MAX_FRAME_LEN" on lmic.h and "MAX_LEN_FRAME" on lorabase.h . Both are part of LMIC library source.

closed 2018-02-20 by javobarra