gotthardp/lorawan-server #308: repeated reset

p-nahrgang, 2018-02-21:
Hi,
I have used the lorawan-server Version 0.4.13 and got "ack_lost" warnings and the payload to the websocket to Node Red was delayed.
But the rest was running fine.

I red the record #241 and upgraded the server to the version 0.5.0.

Now I have some trouble. See debug-log.

I have now the following errors and warnings:
(warning) repeated_reset
(error) tx_freq
(warning) ack_lost

I'm working with Laird Devices RM186 (node) and RG1xx (gateway).

With the version 0.4.13 I had the message "Joining" and "Joined" at the Laird devices.
With the version 0.5.0 I can't achive a Joined.

Any recomendation is highly appreciated.

8-02-21 11:04:38.407 [debug] <0.308.0> Supervisor {<0.308.0>,lorawan_backend_sup} started lorawan_connector_sup:start_link() at pid <0.309.0>
2018-02-21 11:04:38.421 [debug] <0.308.0> Supervisor {<0.308.0>,lorawan_backend_sup} started lorawan_backend_factory:start_link() at pid <0.310.0>
2018-02-21 11:04:38.422 [debug] <0.300.0> Supervisor {<0.300.0>,lorawan_sup} started lorawan_backend_sup:start_link() at pid <0.308.0>
2018-02-21 11:04:38.423 [info] <0.31.0> Application lorawan_server started on node 'lorawan@lorawan-server'
2018-02-21 11:04:48.086 [info] <0.304.0> gateway C0EE40FFFF2934EF {connected,{{79,204,93,171},37948}}
2018-02-21 11:10:09.084 [info] <0.493.0> device 0000000000000004 {join,<<"02FA34F7">>}
2018-02-21 11:10:09.085 [warning] <0.493.0> node 02FA34F7 {repeated_reset,114}
2018-02-21 11:10:09.088 [debug] <0.493.0>@lorawan_mac:encode_accept:392 Join-Accept <<"02FA34F7">>, netid <<0,0,1>>, cflist undefined, rx1droff 1, rx2dr 4, appkey <<"4CC4CD2BB936D40A2A03FD1D9F669CCF">>, appnce <<"257286">>
2018-02-21 11:10:09.088 [debug] <0.493.0>@lorawan_handler:join:107 Join-Accept in RX1: {rxq,868.1,<<"SF7BW125">>,<<"4/5">>,undefined,343636587,-42,8.8}
2018-02-21 11:10:28.169 [warning] <0.304.0> gateway C0EE40FFFF2934EF {ack_lost,50.0}
2018-02-21 11:15:34.322 [info] <0.65.0> alarm_handler: {set,{system_memory_high_watermark,[]}}
2018-02-21 11:21:19.124 [info] <0.769.0> device 0000000000000004 {join,<<"02FA34F7">>}
2018-02-21 11:21:19.125 [warning] <0.769.0> node 02FA34F7 {repeated_reset,115}
2018-02-21 11:21:19.128 [debug] <0.769.0>@lorawan_mac:encode_accept:392 Join-Accept <<"02FA34F7">>, netid <<0,0,1>>, cflist undefined, rx1droff 1, rx2dr 4, appkey <<"4CC4CD2BB936D40A2A03FD1D9F669CCF">>, appnce <<"DF275A">>
2018-02-21 11:21:19.129 [debug] <0.769.0>@lorawan_handler:join:110 Join-Accept in RX2: {rxq,868.3,<<"SF7BW125">>,<<"4/5">>,undefined,1013703339,-41,9.5} {1,4,4}
2018-02-21 11:21:19.148 [error] <0.304.0> node 02FA34F7 <<"tx_freq">>
2018-02-21 11:21:39.200 [info] <0.775.0> device 0000000000000004 {join,<<"02FA34F7">>}
2018-02-21 11:21:39.201 [warning] <0.775.0> node 02FA34F7 {repeated_reset,116}
2018-02-21 11:21:39.203 [debug] <0.775.0>@lorawan_mac:encode_accept:392 Join-Accept <<"02FA34F7">>, netid <<0,0,1>>, cflist undefined, rx1droff 1, rx2dr 4, appkey <<"4CC4CD2BB936D40A2A03FD1D9F669CCF">>, appnce <<"27C075">>
2018-02-21 11:21:39.204 [debug] <0.775.0>@lorawan_handler:join:107 Join-Accept in RX1: {rxq,868.5,<<"SF9BW125">>,<<"4/5">>,undefined,1033737124,-40,9.2}
2018-02-21 11:22:28.408 [warning] <0.304.0> gateway C0EE40FFFF2934EF {ack_lost,100.0}

gotthardp, 2018-02-21:
`repeated_reset` means the device actually never joined
it's caused by the `tx_freq` error which means "requested frequency is not supported by TX RF chain"
sounds like config misalignment between gateway and server. what "chain" did you set in the server's gateway config and what is the gateway's global.conf?

p-nahrgang, 2018-02-23:
**repeated_reset means the device actually never joined**
But on the Node-Page of the Server-GUI there is a "Last Join Entry" of today from the Node!
I'm using the Laird RG1xx "Default-Settings" for Europe in the Gateway. See Screenshots below.

[image]

[image]
[image]

With these Settings the RG1xx gatway communicates with TheThingsNetwork without problems.
Also with the Version 0.4.13, I didn't see these tx_freq-Errors.

In the Server I have selected under Network "EU 863-870". But I saw a strange behavior at this selection box. If I selected a other Region (e.g. US 902-928MHz) I got the error-message "invalid-form" and when I afterwards select again "EU 863-870" - which is the correct one- I got again the message "invalid-form". So I'm not sure if the server works internally with the correct parameter.

I'm trying to get more detailed information from Laird regarding gateway settings (e.g. chain).

I understand that global.conf means e.g. EU_868. I think this is confirmed.

gotthardp, 2018-02-23:
The "chain" is the "0" or "1" in your "Radio 0" or "Radio 1". For example I can see that for 868.3 the "Radio 1" has to be used. The default server config recommends "0", so this may be the misconfiguration.

p-nahrgang, 2018-02-23:
Now I have changed 868.3 to Radio 0 (see Screenshot below), but this didn't solve the problem.

On the server-site I can't change the TX Chain on the gateway side. When I enter "1" and click the "Save changes" button, I get the message "Data has been changed. Please reload and repeat Action. I have repeated this Action several times, but after reload there is always TX Chain "0" displayed.

[image]

gotthardp, 2018-02-23:
The "reload" issue was fixed in 0.5.0 (i believe). It's caused by data sent by the gateway. If you temporarily disconnect your gateway the value should be editable.

gotthardp, 2018-03-01:
Does the problem persist also in 0.5.x? May I close the issue?

p-nahrgang, 2018-03-05:
Sorry for the delayed response. I was out of office for one week.
Now the tx_freq-error is solved. I have changed the TX Chain to "1" and than back to "0" and afterwards the tx_freq-error was disappeared. It seems that the server displayed a "0", but did not work with this "0".
Now I have still:
- ack-lost (warning)
- repeated-resets (warning) and
- timeout during the joining sequence at the node.
Which parameters are related to these problems?
There is one more issue, where I have some difficulties. At the "Backend" I have defined two Connectors and now I would like to define a Handler with one of these connectors, but on the handler side there is the word connector displayed without box (frame) next to the parameter-name.
With the version 0.4.13 I used Node-Red with the Websocket-Link:
ws://127.0.0.1:8080/ws/nodes/[Dev.-Adr.]/raw 
Node-Red displayed with this link "connected".
But with 0.5.0 there is disconnected displayed at the node-red websocket icon.
What do I have to change to get a correct handler and a ws-connect to node-red?

gotthardp, 2018-03-05:
* `ack-lost` is a gateway to server connectivity issue
* `repeated-resets` and `timeout` is again caused by malfunctioning downlink. Are you sure the gateway is now sending the downlinks successfully? Please check the gateway log.
* If you want me to help you debugging the handlers/connectors, please send me screenshots with the settings you are using

p-nahrgang, 2018-03-06:
Thank you very much for your support!
I have increased the values for the Rx1 join delay to 5 and Rx2 join delay to 6 and now the "repeated-reset" disappeared an the Joining sequence is working, but there are still "ack-lost" warnings. In addition I have a big delay in the data transmission.  There is a node-program example from Laird with a timer of 25 sec.. With "TheThingsNetwork" and this program ervery 25 sec. a data Frame (payload) is transmitted. With the lorawan-server and this Laird-example it takes 2min and 40 sec for one payload transmission.
Is there a limitation for the data transmissions in the lorawan-server?
Could it be that there is a relation to the "ack-lost" Problem?

Here the screenshots for the connector/handler-settings:

[image]
[image]
[image]
Any recommendations are appreciated!

gotthardp, 2018-03-06:
First of all, your Connector is miconfigured:
 - You have to assign it to some Application, corresponding to a Handler
 - Instead of the 222222 you should use the {devaddr} template to the uplink frames are correctly matched

There is no rate limitation. It may be because your device is transmiiting on frequencies that are not received by your gateway. Have a look on the device debug console when exactly is the device transmitting and check if the gateway and the server has received the uplink.

gotthardp, 2018-03-06:
Also, a common misconfiguration is that the gateway is configured to use both TTN and my server, which means that only some frames are sent to my server. Check the gateway config there is only one server listed.

p-nahrgang, 2018-03-08:
I have corrected the Connector and now I get a connection to node-red. 
Sometimes the data are passed on to node-red, sometimes not. Now I have to find out why sometimes not.
You wrote:
"Instead of the 222222 you should use the {devaddr} template to the uplink frames are correctly matched"
In the connector-description is stated for the websocket:
Publish Uplinks to a URL pattern starting with a slash, e.g. '/ws/uplink/{devaddr}'
What do you mean with a template? Where can I find an example and a description about it?

Where can I find a recommended frequency-list (channel-list)?
What frequencies are used when Tx-channel are set to 0 or 1 on the server?

Here a screenshot after the following sequence:
- Joining with the TheThingsNetwork
- Changing the Server from TTN to lorawan-server.

[image]

You can see first a shorter interval and later on a longer interval (adr. 26012DF3 and 22222222 are the same node !).
If I understand this correct, then the interval depends on the joining sequence.
There might be a different parameter exchange between TTN and your Server.
Should I make a WireShark-trace in order to analyse this?

In the gateway, there is a function for displaying the traffic. 
With this function I have observed, that there might be a relation between ack_lost and a channel-change at the Gateway.

p-nahrgang, 2018-03-08:
One important information: For TTN and your Server the same channels (frequencies) are used.
At the Laird-Gateway you can't have more than one server-connection at the same time.

gotthardp, 2018-03-08:
ack_lost is a gateway to server connectivity issue; a wireshark may show something

the templates are described here: https://github.com/gotthardp/lorawan-server/blob/master/doc/Connectors.md#patterns

the channel/frequencies are defined in your gateway config. That config defines what radio (tx-channel) uses what frequencies and which is for both uplinks and downlinks

p-nahrgang, 2018-03-09:
Please find attached two wireshark joining-traces with the same settings.

1. RG1xx - your-lorawan-server
2. RG1xx - TTN

For the orientation here some IP-addresses

RD1xx: 192.168.178.30
TNN: 52.169.76.203
your-lorawan-server: 192.168.178.35

[RG1xx-Join.zip](https://github.com/gotthardp/lorawan-server/files/1797091/RG1xx-Join.zip)

You know the detailed meaning of the lorawan-bytes and it might be easy for you to recognize the difference. I'm convinced, the difference is the cause of my problem.

Here the settings of the frequencies in the Laird RG1xx-gateway for both cases the same settings has been used:

SF 0 Radio 1; 868,1 MHz
SF 1 Radio 1; 863,3 MHz
SF 2 Radio 1; 868,5 MHz
SF 3 Radio 0; 867,1 MHz
SF 4 Radio 0; 867,3 MHz
SF 5 Radio 0; 867,5 MHz
SF 6 Radio 0; 867,7 MHz
SF 7 Radio 0; 867,9 MHz

LoRa STD Radio 1; 868,3 MHz
FSK           Radio 1; 868,8 MHz

Thanks in advance for analysing.

p-nahrgang, 2018-03-09:
Here some additional Information, which may help to find the cause of the Problem.
I got a Special program from laird in order to check the available channels.
The sceenshots shows the result (before and after joining):
[image]
[image]
There is no change after joining. In addition you can see a sync-problem during the joining sequence.
In case of TTN, there are more channels available after joining.
Here the screenshots of the Network channel-setting:
[image]
[image]
[image]
Please let me know, if I did something wrong.

gotthardp, 2018-03-09:
Hello. First of all, if you add new channels I guess you should also increase the "Initial Channels" config. By 0-1 you even discard one of the default 3.

gotthardp, 2018-03-09:
Next, the default RX2 frequency is 869.525. I don't see it in your list of frequencies supported by your gateway,

gotthardp, 2018-03-09:
Third, the Wireshark logs show that no UDP traffic is coming back to the gateway, neither for TTN nor for my server. This is a problem for both, but only my server reports that apparently ;-)

p-nahrgang, 2018-03-12:
You wrote:
"Hello. First of all, if you add new channels I guess you should also increase the "Initial Channels" config. By 0-1 you even discard one of the default 3."

Sorry, but this is not correct!

Here is the result of these Settings:

[image]

You wrote:
Next, the default RX2 frequency is 869.525. I don't see it in your list of frequencies supported by your Gateway.

Why is this frequency not displayed in the Networks/ADR-form?
I was using the defauls setting but there is displayed a "1".

Why isn't a Standard lora-frequency (868,1; 8683,1 or 868,5) used for this purpose?

Please don`t unterstand me wrong.
Overall you have developed a very good product and I appreciate this product very much, but there is room for improvements.

The Information with the 869,525 MHz for the RX2 frequency was the key for the solution of my problem. Now I have no longer a delay of the data-transmission.

Never the less, there are still some warnings like ack_lost or very seldom download_lost.

So after I got the answers to my questions (above) we should close this issue.
For the remaining warnings, then I will open a new issue.

Thank you very much for your Support!

gotthardp, 2018-03-12:
1) I don't really understyand what is incorrect. Your previous comment was that you see only 3 channels. If you now sent a screenshot with 7 channels so on the contrary it looks like my suggestion helped. Or you kept 0-1 as your channel list?
2) Where did you see "1" under RX2 Frequency? The [Regions.md](../blob/master/doc/Regions.md) display the correct value. Why this value is used? Because it is described so in the standard; it was not my decision.
3) Seldom downlink_lost will never disappear. In LoRa sometimes a packet is lost due to interferrence or some other RF issue. This message just warns that this has happened.

p-nahrgang, 2018-03-13:
Sorry that I didn`t tell the whole story in detail.
My Problem was too less channels for data transmission due to a problem during the joining-sequence.
The cause of the Problem was the "Initial RX2 Frequency" (Default 869,525 MHz).
So I changed in the lorawanserver-form Networks/ADR/Initial RX2 Freq (MHz) the value from "1" to "868,3" and afterwards I have had more available channels for data-transmission (see screenshots above, displayed by a spacial program on the node).
Even with this Setting Networks/Channels/Initial Channel "0-1" I had beside the standard channels (868,1; 868,3 and 868,5) some additional channels available.
I did see the "1" in the box "Networks/ADR/Initial RX2 Freq (MHz) after the installation of the lorawan-server. At this place the 869,525 MHz should be displayed.
In this box you have to enter a MHz-value and I don't think 1-MHz is correct.
Now I have again changed this entry to "1" and "1" (MHz) was accepted. 
This value should be checked against the allowed lorawan frequency-band.

gotthardp, 2018-03-15:
I just added the value check. The other questions were answered i believe.

closed 2018-03-15 by gotthardp