gotthardp/lorawan-server #309: websocket and {app}

hallard, 2018-02-22:
@gotthardp 
I may doing things wrong but today I started to play with v0.5.1, connectors and handlers
I succeed to do what you told me last time on #288, I created my handlers and connector according the documentation
[image]
then connectors
[image]

and the websocket connector
[image]

Then I tried with to put `/ws/uplink/{devaddr}` in Publish Uplinks  it works, you can see log below

```
2/22/2018 8:28:31 PM RECEIVED: {"all_gw":[{"lsnr":10.5,"mac":"FCC23DFFFE0F44B5","rssi":-41}],"app":"ch2i","appargs":"thethingsnode-02","battery":255,"codr":"4/5","data":"040201CB0502014C06020149","datetime":"2018-02-22T19:28:31Z","datr":"SF7BW125","devaddr":"02D0841F","deveui":"0004A30B001C546F","fcnt":14,"freq":867.1,"lsnr":10.5,"mac":"FCC23DFFFE0F44B5","port":2,"rssi":-41}
2/22/2018 8:23:28 PM RECEIVED: {"all_gw":[{"lsnr":9.5,"mac":"FCC23DFFFE0F44B5","rssi":-45}],"app":"ch2i","appargs":"thethingsnode-02","battery":255,"codr":"4/5","data":"040201C90502014C06020149","datetime":"2018-02-22T19:23:28Z","datr":"SF7BW125","devaddr":"02D0841F","deveui":"0004A30B001C546F","fcnt":13,"freq":867.9,"lsnr":9.5,"mac":"FCC23DFFFE0F44B5","port":2,"rssi":-45}
2/22/2018 8:22:29 PM RECEIVED: {"all_gw":[{"lsnr":9,"mac":"FCC23DFFFE0F44B5","rssi":-37}],"app":"ch2i","appargs":"thethingsnode-02","battery":255,"codr":"4/5","data":"040201CA0502014C0602014907020007","datetime":"2018-02-22T19:22:29Z","datr":"SF7BW125","devaddr":"02D0841F","deveui":"0004A30B001C546F","fcnt":12,"freq":868.3,"lsnr":9,"mac":"FCC23DFFFE0F44B5","port":4,"rssi":-37}
2/22/2018 8:22:05 PM CONNECTED
2/22/2018 8:22:05 PM Connecting to: ws://192.168.1.50:8080/ws/uplink/02D0841F
```

The problem if that when I use `/ws/uplink/{app}` in Publish Uplinks. This is not working, I can't connect to websocket connector. What I'm doing wrong
```
2/22/2018 8:31:49 PM DISCONNECTED
2/22/2018 8:31:49 PM ERROR: undefined
2/22/2018 8:31:49 PM Connecting to: ws://192.168.1.50:8080/ws/uplink/ch2i
2/22/2018 8:31:47 PM DISCONNECTED
```

gotthardp, 2018-02-22:
Looks like a bug. Will you send me the debug.log, please?

hallard, 2018-02-22:
Just sent a PM, thanks

gotthardp, 2018-02-23:
I am sorry. It was a bug. I just fixed that in the latest 'master'.

hallard, 2018-02-23:
Thanks, will try on next release, because I need to rebuild all ;-)

closed 2018-03-01 by gotthardp