gotthardp/lorawan-server #310: SSL only, works  but still process on port 8080

hallard, 2018-02-23:
@gotthardp 

I enabled only SSL config on port 8443, but I still have a process on port 8080

```
cat sys.config
[
% server configuration
{lorawan_server, [
    % UDP port listening for packets from the packet_forwarder Gateway
    {packet_forwarder_listen, [{port, 1680}]},
    % HTTP port for web-administration and REST API
    % {http_admin_listen, [{port, 8080}]},
    % HTTP/SSL port
    {http_admin_listen_ssl, [
        {port, 8443},
        {certfile, "/etc/letsencrypt/live/toto/fullchain.pem"},
        {keyfile, "/etc/letsencrypt/live/toto/privkey.pem"}
    ]},
```
As you can see here, there is a beam process binded on port 8080
```
sudo netstat -tulpn
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 127.0.0.1:3306          0.0.0.0:*               LISTEN      82328/mysqld
tcp        0      0 127.0.0.1:6379          0.0.0.0:*               LISTEN      330/redis-server 12
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      37636/beam.smp
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      37620/nginx -g daem
tcp        0      0 0.0.0.0:4369            0.0.0.0:*               LISTEN      37645/epmd
tcp        0      0 0.0.0.0:1843            0.0.0.0:*               LISTEN      98255/node-red
tcp        0      0 0.0.0.0:8883            0.0.0.0:*               LISTEN      298/mosquitto
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      334/sshd
tcp        0      0 0.0.0.0:1880            0.0.0.0:*               LISTEN      37620/nginx -g daem
tcp        0      0 127.0.0.1:25            0.0.0.0:*               LISTEN      83441/exim4
tcp        0      0 0.0.0.0:8443            0.0.0.0:*               LISTEN      37636/beam.smp
tcp        0      0 0.0.0.0:50427           0.0.0.0:*               LISTEN      37636/beam.smp
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      37620/nginx -g daem
tcp        0      0 0.0.0.0:1883            0.0.0.0:*               LISTEN      298/mosquitto
```
And this make another process on my server not starting on port 8080

Any idea ?

Thanks

gotthardp, 2018-02-23:
Try to keep (uncomment) the settings but set it to
`{http_admin_listen, undefined}`

hallard, 2018-02-23:
Yeah, did the trick, thanks ;-)

closed 2018-02-23 by hallard