gotthardp/lorawan-server #311: no ACK received for confirmed Tx

vascode, 2018-02-23:
Hi 

v0.5.1 is running and my data from end-device arrives the server but ACK is not received on the end-device nor on gateway. 

I noticed "wrong_data" appeared in the event list. I'm wondering if my setup is problematic or the data I am sending is. Can you tell me what that message is about?

error | 2018-02-23 16:34:54 | 2018-02-23 16:34:54 | 1 | node | 00000150 | wrong_data | {1,<<"hello">>}
-- | -- | -- | -- | -- | -- | -- | --

gotthardp, 2018-02-23:
The server is configured to use the *semtech-mote* application, but your device sent some other data. Make sure your firmware uses the [supported format](https://github.com/Lora-net/LoRaMac-node/blob/master/src/apps/LoRaMac/classA/LoRaMote/main.c#L207).

In the upcoming release this error will be renamed to 'not_semtech_mote'.

vascode, 2018-02-27:
Two questions here : 
1. What happens when the server receives a not supported format of data. Does it still show it on the "Received Frames" but it doesn't send down ACK to the end-device? 

2. This is the data I have played with Semtech server before and worked before. But I still get a wrong_data. Does the server just care about the length of data (16 bytes) or anything else? Do you see a problem in this data?
"\00\00\00\07\00\00\80\00\3A\6D\4B\C6\07\EC\00\85"

gotthardp, 2018-02-27:
1) It depends on the application. When the semtech application receives unsupported frame, it is just logged and then ignored. No ACK is sent.

2) The data is OK, but the port could be wrong too. Semtech uses Port "2". Data on every other port are also treated as unexpected.

vascode, 2018-02-27:
Port "2" works. But I noticed that my end-device received "\01" payload instead of just ACK for the response of my confirmed data message.

gotthardp, 2018-02-27:
That is what the application does-- it sends a 1-byte command to blink with the LED.

jot4p, 2018-03-01:
It's possible to add my own device json settings ?

I need to parse this:
not_semtech_mote | {5, <<"{\"type\":\"meteo\",\"t\":8.1,\"hr\":100.0,\"pa\":97993.0,\"uv\":0,\"ane\":0.0}">>}

gotthardp, 2018-03-01:
Please always open a new issue for new questions. What binary data does you device send? What you put here is not syntactically valid.

jot4p, 2018-03-01:
My device send json directly:

payload:   {"type":"meteo","t":8.1,"hr":100.0,"pa":97993.0,"uv":0,"ane":0.0}

jot4p, 2018-03-01:
raw frame:
7B2274797065223A226169722D7061727469636C6573222C22706D31223A2D312E30302C22706D325F35223A2D312E30302C22706D3130223A2D312E30307D

gotthardp, 2018-03-01:
That's sooooo inefficient. The device is wasting RF resources. Can you can it to something binary? Sending JSON in LoRa frames feels crazy I have to say.

jot4p, 2018-03-01:
Yes.. I use this on my new devices.. but my old devices send JSON it's possible to integrate ?

gotthardp, 2018-03-01:
Write an own Parse Uplink function. Something like:
```erlang
fun(Fields, Payload) ->
  jsx:decode(Payload, [return_maps, {labels, atom}])
end.
```

gotthardp, 2018-03-01:
I believe the questions were answered. Feel free to open new issue if you encounter some troubles.

closed 2018-03-01 by gotthardp