gotthardp/lorawan-server #313: String payload

webfrank, 2018-02-25:
I would like to receive in JSON a string payload but I don't know how to write fun parser, I used something like this but it doesn't work:

fun(Fields, <<Rest/binary>>) ->
  Fields#{data => Rest}
end.

gotthardp, 2018-02-25:
First of all, what is a "string payload"? ASCII characters directly in the uplink message like <<"HELLO">>?

webfrank, 2018-02-25:
I was sending as a payload a string "SOS", but I think I solved it using Rest/bitstring and NOT using data as JSON variable, I changed it in msg because I think it is override with base64 decoded data

gotthardp, 2018-02-25:
The "data" field gets auto-encoded to hex-string, but you may assign the payload to some other field:
```erlang
fun(Fields, <<Rest/binary>>) ->
   Fields#{payload => Rest}
end.
```

gotthardp, 2018-03-01:
Just so you know: I just added a "ASCII Text" payload parser, so with the latest server you can use that instead of writing that simple parse function.

closed 2018-03-01 by gotthardp