gotthardp/lorawan-server #314: Data reporting

minelpi, 2018-02-26:
Hi,
Thanks for your kind contribution, it made me able to make my first IoT system after 4 years phd study :-) on the topic.
One question, when I am sending 'hi', the data column of the received frames shows '6869' which is the hex presentation of the received data. Could you make an option to convert this data to ascci or etc?

G.L.

gotthardp, 2018-02-26:
This can be achieved by writing a Parse Uplink function that will simply store the payload to some other field that doesn't get auto-converted
```erlang
fun(Fields, Payload) ->
  Fields#{payload => Payload}
end.
```

minelpi, 2018-02-26:
Thanks for answer and support.
Using that part of code, It shows me an error event:

error | 2018-02-26 13:36:40 | 2018-02-26 13:36:40 | 1 | node | 26011939 | unexpected_data | {rxdata,13,1,<<"hi">>,false,true}
-- | -- | -- | -- | -- | -- | -- | --

gotthardp, 2018-02-26:
This looks like you switched to to the semtech-mote application. You should define your own backend handler and connector and then use this as your application.

minelpi, 2018-02-26:
yes.
Thanks.
Solved.

closed 2018-02-26 by gotthardp