gotthardp/lorawan-server #316: no match of right hand value

stefanoridolfi, 2018-02-27:
Hi,
I installed server and Erlang/OTP in centos 7.I create profile, device etc. 
When motes try to connect I saw this message in log
@lorawan_mac:ingest_frame:23 gen_statem <0.450.0> in state idle terminated with reason: no match of right hand value {aborted,{badarg,[{crypto,strong_rand_bytes_nif,[-1],[]},{crypto,strong_rand_bytes,1,[{file,"crypto.erl"},{line,327}]},{lorawan_mac,create_devaddr,3,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,125}]},{lorawan_mac,handle_join,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,107}]},{mnesia_tm,apply_fun,3,[{file,"mnesia_tm.erl"},{line,836}]},{mnesia_tm,execute_transaction,...},...]}} in lorawan_mac:ingest_frame/1 line 23
Thanks
Rgds

gotthardp, 2018-02-27:
Hi. I suspect your network SubID is too long.

The server should check it, that's my fault, but you shouldn't set so long number too ;-)

stefanoridolfi, 2018-02-27:
Hi,
many thanks for wery fast answer, I changed value as you suggested, it work
:-)

Regards

Stefano

gotthardp, 2018-02-27:
I am glad it works :)

closed 2018-02-27 by gotthardp