gotthardp/lorawan-server #317: ADR rssi

gnob417, 2018-02-28:
Hi,
there is 

>  % receiver sensitivity for maximal DR in all regions is -120 dBm, we try to stay at -100 dBm
in ADR codes.

I want to do experiment in indoor environment with some obstacles. 
so i have to correct a little for nodes to stay at the value more reliable than -100dBm.

maybe in the below code, i have to change -96 and -102 value??

```
TxPower2 = if
            AvgRSSI > -96, TxPower < MinPower ->
                PwrStepUp = trunc((AvgRSSI+100)/4), 
                % there are 2 dB between levels, go slower
                lager:debug("Power ~s: average rssi ~w, power ~w -> up by ~w",
                    [lorawan_utils:binary_to_hex(Node#node.devaddr), round(AvgRSSI), TxPower, PwrStepUp]),
                min(MinPower, TxPower+PwrStepUp);
            AvgRSSI < -102, TxPower > MaxPower ->
                PwrStepDown = trunc((AvgRSSI+98)/2), % go faster
                lager:debug("Power ~s: average rssi ~w, power ~w -> down by ~w",
                    [lorawan_utils:binary_to_hex(Node#node.devaddr), round(AvgRSSI), TxPower, PwrStepDown]),
                max(MaxPower, TxPower+PwrStepDown); % steps are negative
            true ->
                TxPower
        end,
```

gotthardp, 2018-02-28:
Yes, that is correct.

closed 2018-03-01 by gotthardp

gnob417, 2018-03-05:
so if i wnat nodes to stay at the value more reliable than -100dBm (let's say -90dBm),
what values should be changed and how much?

as i mentioned above, only -96 and -102 values have to be modified? (like -86 and -92)

gotthardp, 2018-03-05:
You want to increase the hysteresis or stay at -90dBm instead of -100?

Anyway, the algorithm is pretty straightforward so you can change it and experiment as you like.