gotthardp/lorawan-server #319: Help with frame parser

jot4p, 2018-03-02:
Hello, 

I need help with frame parser.

my configuration is: 
https://imgur.com/a/ZZfX6 

```
fun(Fields, Payload) ->
  jsx:decode(Payload, [return_maps])
end. 
```

But my frame is partitioned... 

https://imgur.com/a/mRrSZ 

I need help to convert from new to old frame in MQTT. 

Thanks.

gotthardp, 2018-03-02:
I am not sure I understand. What is "old" and "new" frame?

jot4p, 2018-03-02:
New is the frame sended from your server .. old is the frame sended directly from my gateway+node red Json parser

gotthardp, 2018-03-02:
I see. You will need to adapt your application to the new JSON structure. This is beyond the lorawan-server.

jot4p, 2018-03-02:
But, here for example: https://imgur.com/KDliHht i have the DevAddr, Mac, Signal and frame data in the same line.. it's possible to use all this entries in my handler ?

gotthardp, 2018-03-02:
I see. If you want those parameters, you need a slightly more complicated fun. Try:
```erlang
fun(Fields, Payload) ->
  maps:merge(Fields, jsx:decode(Payload, [return_maps]))
end. 
```

jot4p, 2018-03-05:
Works. Thanks.

closed 2018-03-05 by gotthardp