gotthardp/lorawan-server #320: Libelium Smart Parking Sensor and Lorawan-server

jot4p, 2018-03-02:
Hello, 

I have some smart parking sensors:
http://www.libelium.com/downloads/documentation/smart_parking_technical_guide.pdf

this devices, after version 2 of firmware requires a callback server.

> Libelium provides the source files of a simple web application to deploy in your server. This remote node
> configuration web application provided by Libelium will receive Sigfox and LoRaWAN requests, sending a response
> back with the Smart Parking node configuration.

The programming software just asks (OTAA) DeviceEUI, PORT, AppKey, AppEUI

When I start the device (or devices) the log error of server is:

2018-03-02 11:34:39.643 [error] <0.3853.0> device 00800000A000XXXX unknown_deveui
2018-03-02 11:34:45.964 [error] <0.3856.0> device 00800000A000XXXX unknown_deveui
2018-03-02 11:34:52.278 [error] <0.3858.0> device 00800000A000XXXX unknown_deveui
2018-03-02 11:35:01.455 [error] <0.3861.0> device 00800000A000XXXX unknown_deveui
2018-03-02 11:35:07.920 [error] <0.3863.0> device 00800000A000XXXX unknown_deveui
2018-03-02 11:35:14.384 [error] <0.3865.0> device 00800000A000XXXX unknown_deveui
2018-03-02 11:36:27.240 [error] <0.4017.0> device 00800000A000XXXX unknown_deveui
2018-03-02 11:36:33.563 [error] <0.4021.0> device 00800000A000XXXX unknown_deveui

With my Multitech Conduit gateway and node-red (installed on conduit) i receive:
 
https://imgur.com/a/PFXBz 

Parsing the frame to json:
https://imgur.com/a/UszqZ

what is the problem with server ?

gotthardp, 2018-03-02:
This message was not sent by my server. You are using some other LoRaWAN server, which I don't understand and thus cannot support.

jot4p, 2018-03-02:
No, is sended for your server, the used images are prints from my conduit gateway in order to help to understand package

gotthardp, 2018-03-02:
Could you send me screenshot of the Handler settings and the MQTT/JSON messages the server produced, please? The JSON structures in the screenshots above are clearly not produced by my server (the EUI has different format and some _msgid is added)

jot4p, 2018-03-02:
Ok
this ?  https://imgur.com/a/rinT6

gotthardp, 2018-03-02:
If you see unknown_deveui, then your device is not configured in the server. For such devices no uplinks are sent anywhere. You should add the device in the "Commissioned" table.

jot4p, 2018-03-02:
Oh.. ok.. I will test...

jot4p, 2018-03-05:
The device is detected now, but i need to send callback response like this (page 68)
http://www.libelium.com/downloads/documentation/plug_and_sense_smart_parking_technical_guide.pdf

It's possible with your server ?

gotthardp, 2018-03-05:
The documentation talks about generic callbacks sending uplink data to a server. Yes, that can be done with my server too.

jot4p, 2018-03-05:
Ok..
I need help with this if it's possible.

My callback server receive: 
`127.0.0.1 - - [05/Mar/2018:17:58:38 +0000] "POST /uplink/01DDA4ED/ HTTP/1.1" 404 1054 "-" "-"`

The parser for "actility" is something like:
`<?php
bla bla bla
try {
   ... bla bla bla.. 

    $data = json_decode(file_get_contents('php://input'), true);
    if (array_key_exists('DevEUI_uplink', $data)) {
        $log->addInfo("Received info: " . print_r($data, true));
        $data = $data['DevEUI_uplink'];

        $frame = new Frame($data['DevEUI']);
        $data_frame = $frame->get_info($data["payload_hex"]);

        //cURL to send payload.
        if ($frame->waiting_response()) {
            $server_url = $config->get('my_service', 'server_url', '');
            $params = array(
                "DevEUI" => urlencode($data['DevEUI']),
                "FPort" => urlencode($data['FPort']),
                "Payload" => urlencode($frame->get_response())
            );

            $ch = curl_init();
            $post = http_build_query($params);
            $res = exec('curl -H "Content-type:application/x-www-form-urlencoded" -X POST "'.$server_url.'?'.$post.'"');
            $log->addInfo('Information sent');
            $log->addInfo('cURL executed: '.$server_url.'?'.$post);
            $log->addInfo("cURL output: " . print_r($res, true));

        } else {
            $log->addInfo("No information to send.");
        }

        //To-Do: get the information received: $frame->get_data();
    }
} catch (Exception $e) {
    $log->addError($e->getMessage());
}`

I need to convert this for your server..

gotthardp, 2018-03-05:
This looks like a PHP receiving one HTTP request, then parsing the data and sending another HTTP someplace else? Am I right?
You should start by reviewing 1) what the sensor sends and 2) what the target application expects. Then I can help you with the erlang parser function that will turn 1) into 2).

jot4p, 2018-03-05:
Yes, this is part of libelium callback server .. Is required for Smart Parking sensors. 

Its necessary get frame from node to lorawan server (your cose) and send from server to callback server (libelium) and this server contact libelium api to validate the node frame, after that callback server send an “ok” response to node with your lorawan server.

gotthardp, 2018-03-05:
I have no time to read the documentation. If you want me to help you, you should tell me exactly what is the binay frame sent (1) and what is the HTTP request expected (2).

jot4p, 2018-03-06:
ok.

I need to respond to 3 types of frames:

1 - Response to the Start Frame number 1, transmitted by the node in the activation process. This response frame allows the initial time synchronization of the internal clock. In normal conditions, the activation process happens only once.
2 - Response to the Daily Update Frame with a timestamp, in order to perform a time synchronization of the internal clock. This will happen once per day and helps to keep the node’s time drift to minimum.
3- Response to the Daily Update Frame with a new configuration for the node. This will happen very few times, a change in the Smart Parking node configuration will only happen when the administrator of the network decides to change the initial configuration parameters loaded via the USB Programmer.

Theoretically from the server side just need to be sent the "payload"..

jot4p, 2018-03-06:
ok.
I have received the frame from your server:
`127.0.0.1 - - [06/Mar/2018:12:28:54 +0000] "POST /services/i9p/?deveui=0004A30B001BB5EA&payload=070100001DF6F2E25C19DB HTTP/1.1" 200 - "-" "-"`

After processing the information on my callback server I have:

`[2018-03-06 13:28:45.495] [INFO] I9p - Params to send: Array
(
    [deveui] => 0004A30B001XXXXX
    [payload] => 120306020D1C0000
)`

Now, I need to send this frame for your server, and after that, server send this for node. It's possible ?

my code to send from callback to your server:
`
 $params = array(
                "deveui" => urlencode($data['deveui']),
                "payload" => urlencode($frame->get_response()) //get correct response to send
            );
$server_url = $config->get('myservice', 'server_url', '');
			$options = array(
			  'http' => array(
				'header'  => "Content-type: application/x-www-form-urlencoded\r\n",
				'method'  => 'POST',
				'content' => http_build_query($params),
			  ),
			);
			$context  = stream_context_create($options);
			$result = file_get_contents($server_url, false, $context);
			var_dump($result);
`

But on your server I have this: https://imgur.com/a/gYelK 

What is the problem ? uplinks ? downlinks ?

gotthardp, 2018-03-06:
You are almost there. The *Uplink Fields* you listed are automatically included in the Fields variable. The Parse Uplink is to parse the data, i.e. <<8,9,0,0,30,0,137,235,77,7,203>> and set additional fields based on these values.

If you don't need any extra fileds (and pure data is enough), don't set any Uplink Parse function. If you want the data to be parsed, define a pattern that matches the binary you are getting, i.e.
```erlang
fun(Fields, <<A,B,C,D:16,E:16,F:32>>) ->
  Fields#{a => A, b => B, c => C, e => E, f => F}
end. 
```
will match <<8,9,0,0,30,0,137,235,77,7,203>>, assuming A,B,C is 8-bit, D,E is 16-bit and F is 32-bit. You can set the pattern as you like. This will add additional fields to your message with the values sent in the binary.

jot4p, 2018-03-06:
Ok I will try!! 
How

gotthardp, 2018-03-06:
send this received frame to a node? what do you mean? it was just received from the node, why would you send it back?

jot4p, 2018-03-06:
No, this frame is obtained from my callback server

gotthardp, 2018-03-06:
You mean how to send something that your callback server created back to the node? This is called downlink. You can use the same websocket: send a JSON with {"data":"somehexadata"}, or use a Build Downlink function to build the data field from other custom fields. There is an extensive documentation on this subject.

jot4p, 2018-03-06:
Yes, I need to build downlink to resend this frame for my node but earlang is not easy :)

gotthardp, 2018-03-06:
Erlang is easy. Just start with something and I will correct the mistakes :)

jot4p, 2018-03-07:
what is this > "[dev_status_req]" ?

`
2018-03-07 10:12:28.985 [debug] <0.31993.0>@lorawan_mac_commands:build_fopts:73 01DDA4ED <- [dev_status_req]
`

gotthardp, 2018-03-07:
DevStatusReq is a request from the server for device battery status and RX signal indicator. The retrieved values appear under Node Status.

jot4p, 2018-03-07:
I send this info for callback with handler:
`127.0.0.1 - - [07/Mar/2018:18:01:20 +0000] "POST /services/i9p/?deveui=0004A30B001BXXXX&payload=800C190A06F4DC060800BB HTTP/1.1" 200 - "-" "-"`

My callback server log return:
[2018-03-07 18:56:16.642] [INFO] I9p - Params to send: Array
(
    [deveui] => 0004A30B001BXXXX
    [port] => 4
    [payload] => 0000000000000000
)

your server debug return:

`
2018-03-07 17:53:50.601 [debug] <0.10824.0>@lorawan_mac_commands:build_fopts:73 01DDA4ED <- [dev_status_req]
2018-03-07 17:53:55.799 [debug] <0.10828.0>@lorawan_mac_commands:build_fopts:73 01DDA4ED <- [dev_status_req]
2018-03-07 17:54:00.871 [debug] <0.10830.0>@lorawan_mac_commands:build_fopts:73 01DDA4ED <- [dev_status_req]
2018-03-07 17:54:05.934 [debug] <0.10833.0>@lorawan_mac_commands:build_fopts:73 01DDA4ED <- [dev_status_req]
2018-03-07 17:54:10.151 [debug] <0.10834.0>@lorawan_mac_commands:build_fopts:73 01DDA4ED <- [dev_status_req]
2018-03-07 17:54:14.551 [debug] <0.10838.0>@lorawan_mac_commands:store_actual_adr:138 DataRate 01DDA4ED switched to dr 2
2018-03-07 17:54:14.551 [debug] <0.10838.0>@lorawan_mac_commands:build_fopts:73 01DDA4ED <- [dev_status_req]
2018-03-07 17:55:10.996 [debug] <0.10857.0>@lorawan_mac_commands:build_fopts:73 01DDA4ED <- [dev_status_req]
2018-03-07 17:55:22.306 [debug] <0.10860.0>@lorawan_mac_commands:handle_fopts:16 01DDA4ED -> [{dev_status_ans,255,-1}]
2018-03-07 17:55:22.306 [debug] <0.10860.0>@lorawan_mac_commands:store_actual_adr:138 DataRate 01DDA4ED switched to dr 1
2018-03-07 17:55:22.306 [debug] <0.10860.0>@lorawan_mac_commands:handle_status:218 DevStatus: battery 255, margin: -1 (max -17.5)`

this is the response from my node ? What do you think about this ?

Thanks for your help.

jot4p, 2018-03-08:
Hello, 

Other devices with same configurations dosen't work because only try DR 3 ... why ? 

```
2018-03-08 11:31:21.121 [debug] <0.592.0>@lorawan_mac_commands:store_actual_adr:138 DataRate 01393E60 switched to dr 3
2018-03-08 11:31:21.121 [debug] <0.592.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:31:26.262 [debug] <0.596.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:31:31.264 [debug] <0.601.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:31:36.285 [debug] <0.602.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:31:41.272 [debug] <0.607.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:31:46.284 [debug] <0.608.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:31:51.302 [debug] <0.613.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:31:56.301 [debug] <0.614.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:32:01.299 [debug] <0.619.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:32:06.300 [debug] <0.645.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:32:11.312 [debug] <0.650.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:32:16.315 [debug] <0.651.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
2018-03-08 11:37:38.308 [debug] <0.833.0>@lorawan_mac_commands:build_fopts:73 01393E60 <- [dev_status_req]
```

gotthardp, 2018-03-08:
This is a device configuration issue. The server switched the device to a DR 3 and since then the device cannot hear the downlinks. A device log may help.

jot4p, 2018-03-12:
Hi, 

I dont understand.  I need to post my callback info into sever but it's not possible to create valid connection:

My curl request:

```
define('USERNAME','test');
define('PASSWORD','test');
			
curl_setopt($ch, CURLOPT_HTTPHEADER, array('Content-Type: application/json'));
curl_setopt($ch, CURLOPT_CUSTOMREQUEST, "POST");
curl_setopt($ch, CURLOPT_HEADER, 1);
curl_setopt($ch, CURLOPT_USERPWD, USERNAME . ":" . PASSWORD);
curl_setopt($ch, CURLOPT_RETURNTRANSFER, true);
curl_setopt($ch, CURLOPT_POSTFIELDS, json_encode($params));
curl_setopt($ch, CURLOPT_FOLLOWLOCATION, 1);			
curl_setopt($ch, CURLOPT_HTTPAUTH, CURLAUTH_DIGEST); 				
		
$response = curl_exec($ch);
			
$httpCode = curl_getinfo($ch , CURLINFO_HTTP_CODE); // this results 0 every time
			
if ($response === false) $response = curl_error($ch);
//echo stripslashes($response);
$log->addInfo("Response Status Code: ". $httpCode);
$log->addInfo("ERROR: " . stripslashes($response));
			
curl_close($ch);
```
but request return:
```
2018-03-12 12:01:20.981] [INFO] SERVICE - Response Status Code: 204
[2018-03-12 12:01:20.985] [INFO] SERVICE - ERROR: HTTP/1.1 401 Unauthorized
content-length: 0
date: Mon, 12 Mar 2018 11:01:20 GMT
server: Cowboy
www-authenticate: Digest realm="lorawan-server", nonce="f2366c6de1e3e76c62765261ea8de71b"

HTTP/1.1 204 No Content
content-security-policy: default-src * data: 'unsafe-inline' 'unsafe-eval';
content-type: text/html
date: Mon, 12 Mar 2018 11:01:20 GMT
server: Cowboy
```

I need to use this user and password:
https://imgur.com/RF8sNvZ 

But my User/password post is based on my server user (admin,admin) 

How it's possible to send POST for callback handler with authentication like image above ??

jot4p, 2018-03-12:
my curl log:

```
* Hostname localhost was found in DNS cache
*   Trying ::1...
*   Trying 127.0.0.1...
* Connected to localhost (127.0.0.1) port 8080 (#0)
> POST /callback/018F0C23 HTTP/1.1
Host: localhost:8080
Accept: */*
Authorization: Digest username="callback", realm="lorawan-server", nonce="60b5dc6061c86c30e82735223a389e93", uri="http://localhost:8080/callback/018F0C23", response="14f85f8a24933b741ac8668fbb36e840"
Content-Type:application/json
Content-Length: 52

* upload completely sent off: 52 out of 52 bytes
< HTTP/1.1 401 Unauthorized
< content-length: 0
< date: Mon, 12 Mar 2018 18:06:23 GMT
< server: Cowboy
< www-authenticate: Digest realm="lorawan-server", nonce="45506d566847a0a78df1cb0ae9449176"
< 
* Connection #0 to host localhost left intact

```

gotthardp, 2018-03-12:
The log above shows you are getting "HTTP/1.1 204 No Content". This is a valid and correct answer.

I am not a PHP expert. If you capture the traffic via Wireshark, I can have a look though.

jot4p, 2018-03-12:
Yes, but the server return allways 401.. the username and password are always required but I don't need!! 
This password and user are referring to server User (admin/admin) not server "handler connection" user and password like MQTT connection. 

How it's possible to show a correct POST in the server ?  If you try to POST something like:

```curl -H "Content-Type: application/json" -X POST http://localhost:8080/callback/AAAAAAA -d "{\"data\":\"ABCDEFG\"}"
```
The server does nothing. Problem with uplink ?

closed 2018-03-14 by jot4p