gotthardp/lorawan-server #321: Some issues reports

Quentintin, 2018-03-02:
Hello,

I would like to report some bugs to fix.

- Why can we set profile to Commisionned devices and set another profile to Activated (nodes). Is it normal behaviour ?

- When I edit a profile to rename it, it doesn't replace name, but create a new profile and keep the old one.

- Edit `description` field in Gateway doesn't work most of time (but sometimes it works).
> Data has changed. Please reload and repeat the action

- Request to API `localhost:8080/api/gateways` works fine, but we don't have field `description` and `group`. I need it to identify my gateway

Thanks for details

gotthardp, 2018-03-02:
Hello. Some of the bugs are features ;-)
 * The Commissioned device setting is used during JOIN only. Upon join the values are written into the Activated node setting. So you could see Activated node as the current settings and Commissioned node as the Settings to be used after the next reset/join.
 * Yes, this is how the server behaves. You cannot rename: you have to create a new with a different name and then delete the old one.
 * Yeah, you need to be fast when changing the Gateway config. I agree this may be a problem.
 * I don't understand what is the last issue. You can always identify the gateway using MAC, or you can write some description. You'd need some other field?

Quentintin, 2018-03-02:
Thanks for this answvers !

For the last point, I just mean that API Rest don't provide `description` or `group` attribute in JSON response, the same one that I wrote into web UI forms. And i would like to get it with API. 

EDIT : Oh sorry, I didn't see attribute `desc` into JSON response. It was hidden between the 2 big arrays `delays` and `dwell` ! So all is good for me 👍

closed 2018-03-02 by Quentintin