gotthardp/lorawan-server #322: Log or Handler for Unknown frame

hallard, 2018-03-05:
Would it be easy to log Unknown frame in a file (or even best have a handler for it)?
I mean by Unknown frame a valid LoRa frame, with correct SF, payload, Freq, ... good CRC but not a valid LoRaWAN frame.
Some devices and/or application may need to see them (yes one of my need it)

I know it's off topic, but it could be really helpfull and add some feature to this server, and I'm pretty sure it's do able. Packet Forwarder already forward them.

Thanks

gotthardp, 2018-03-05:
Hello. The frames are encrypted and without knowing the frame structure the server cannot decrypt them. I suspect you have some proprietary frame, right?

hallard, 2018-03-05:
Yeah I don't want the server to decrypt them (for now they are not encrypted), just need to have the frame data and attribute (ts, rssi, ...)

gotthardp, 2018-03-05:
All the handlers are related to a DevAddr and a Profile. Without knowing the DevAddr I can only log everything in the log file.

hallard, 2018-03-05:
Oh, got it, I understand, log file is fine ;-)

gotthardp, 2018-03-05:
That can be done

hallard, 2018-03-05:
great, thanks