gotthardp/lorawan-server #323: MQTT - Ignoring Join Frames.

jot4p, 2018-03-05:
Hello, 

It's possible to ignore on my handler Join frames ? 

Thanks

gotthardp, 2018-03-05:
To ignore means that the handler will prevent the server from sending an Accept frame, or to skip the Join Event?

jot4p, 2018-03-05:
I have an handler created to send all received frames from nodes and send this frames to mqtt ..  but i need to ignore/skip join frames on my mqtt 
`{
  "datetime": "2018-03-05T10:39:56Z",
  "devaddr": "E0CBD10E",
  "deveui": "0004A30B001FAB50"
}`

gotthardp, 2018-03-05:
And do you need other Events? If you leave "Publish Events" of your connector empty, you will not be getting any Events.

jot4p, 2018-03-05:
I need to receive all frames like, 

`{
  "datetime": "2018-03-05T11:23:36Z",
  "devaddr": "xxx",
  "deveui": "xxx",
  "port": 3,
  "rssi": -78,
  "bat": 96,
  "co": 0.01,
  "hr": 49.8,
  "no2": 26.436,
  "o3": 0.011,
  "pa": 97993.3,
  "t": 22.6,
  "type": "air-particles"
}`

But i need to ignore join frames..

gotthardp, 2018-03-05:
Then clear the "Publish Events" setting of your connector.

gotthardp, 2018-03-30:
Did I answer your question? May I close the issue?

closed 2018-04-05 by gotthardp