gotthardp/lorawan-server #324: Application and Handlers problem

jot4p, 2018-03-05:
Hi, 

I replace the name of my handlers (FrameParser >> to GenericFrameParser)... after that my devices return an error:
`

unknown_application | <<"FrameParser">>
-- | --
`
I need to change my application name in my profile but the application text box is empty. 

Any solution to clear cache ? or other solution ?

gotthardp, 2018-03-05:
Yeah, this happens when you delete a handler that was referenced in a profile. You should update the profile before deleting the old handler.

xypron, 2018-12-09:
To ensure referential integrity of the database it should not be allowed to delete any object that is still referenced:

* do not allow deleting profiles if used in active or commissioned devices
* do not allow deleting applications if used in connectors or profiles
* and the like

Do not use names as references but immutable IDs. This way renaming an object will not destroy the referential integrity (cf. #324).